prusa3d/PrusaSlicer #3141: Holes doesn't appear

r0xANDt0l, 2019-11-05:
### Version
2.1.0
### Operating system type + version
Windows 10 build  1809

### 3D printer brand / version + firmware version (if known)
Ender 3 with stock board and Marlin 2.0
### Behavior
So, i'm triying to slice a model with holes but it doesn't appear the holes
* _Expected Results_
[image]
* _Actual Results_
[image]

_Is this a new feature request?_
no
#### Project File (.3MF) where problem occurs
[EuroProblem.zip](https://github.com/prusa3d/PrusaSlicer/files/3811493/EuroProblem.zip)

r0xANDt0l, 2019-11-05:
I actually restarted PrusaSlicer but it still doesn't work

Jebtrix, 2019-11-06:
v2.1.0 has issues with duplicate vertices. This does slice in latest git master but result is not optimal.

[EUR-Sorter-topboard_fixed.zip](https://github.com/prusa3d/PrusaSlicer/files/3815643/EUR-Sorter-topboard_fixed.zip)

@bubnikv I know work is being done here but just wanted you to take note of this result comparison.
## Slice with dup vertices merged
[image]

## Slice from latest git master (model as-is)
[image]

closed 2022-06-13 by r0xANDt0l