prusa3d/PrusaSlicer #3142: Feature Request: Modify existing 3mf files instead of overwriting them

Lichtjaeger, 2019-11-06:
Hi,

By its nature, 3mf should be extensible. If you overwrite the file each time you save the project you will also delete modifications made by other programs/platforms.

A concrete example: I added a thumbnail image as a preview to my project but after open and saving with PrusaSlicer this modification was completely gone.

Another example: PrusaSlicer does not need textures but there are full-color printers out there. Shouldn't the 3mf contain the textures for full-color as well as the multi-color definitions of Prusaslicer when possible?

In my opinion, 3mf files would work best as projects if each software would only add and later edit its settings without affecting the settings of other software.

bubnikv, 2021-01-13:
Intended behavior of 3MF when resaving is described in the 3MF format specification

bubnikv, 2021-01-13:
https://github.com/3MFConsortium/spec_core/blob/master/3MF%20Core%20Specification.md

which is built upon 
https://en.wikipedia.org/wiki/Open_Packaging_Conventions

I would have to dig deeper, but I think that documents that are not referenced by OPC are not to be retained.

closed 2021-01-13 by Lichtjaeger