prusa3d/PrusaSlicer #3143: Dual Cursors

impguard, 2019-11-06:
2.1.0+win64

Is it normal to have two cursors show up? When I open the app it all looks good but dragging any file into the window causes two mouse cursors to show up. I've reinstalled the app on two different occasions but this strange bug still shows up.

lukasmatena, 2019-11-15:
Thanks for the report. Does the application freeze for a moment when that happens? Does the second cursor disappear later? Could you please make a short video?

Also, could you provide information about your system (PrusaSlicer->Help->System Info)?

impguard, 2019-11-16:
This is a fresh install of slicer. The application doesn't freeze. I simply use it as normal and a second cursor appears in the main window where I manipulate the print.

The cursor only seems to appear after I drag an STL file in the window. After that, it follows my cursor around as a double. When I select the print itself and move it on the plate, the second cursor seems laggy, but still follows mine.

I'll try to make a video when I get to my desk. I previously uninstalled PrusaSlicer due to the annoyance, but tried it again recently and it occurs immediately without any setting changes.

bubnikv, 2020-03-15:
no response, nobody else complained about it. closing.

closed 2020-03-15 by bubnikv