prusa3d/PrusaSlicer #3145: Feature request. Scripts for when model is loaded

bman2000, 2019-11-07:
2.1Version
Add the ability to run scripts on loaded model such as scale, rotate etc. Such that needed actions can be performed automatically for model preparation and slicing.

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2020-03-15:
@kocikdav is working on a feature, where you would be able to send events from a command line slicer to a running GUI slicer instance. That could satisfy your needs.

rtyr, 2022-02-22:
Implemented. Closing.

closed 2022-02-22 by rtyr