prusa3d/PrusaSlicer #3146: Prusa Slicer crashes when i try to slice this project

fliegermichl, 2019-11-07:
### Version
_PrusaSlicer-2.1.0+win64-201909160915_

### Operating system type + version
_Windows 10 64 Bit german_

### 3D printer brand / version + firmware version (if known)
_Prusa MK3S + MMU2S_

### Behavior
* _crash if i try to slice this project_
#### Project File (.3MF) where problem occurs

[F4U_engine_cowling_L.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3820055/F4U_engine_cowling_L.3mf.zip)

BirkBinnard, 2019-11-07:
Yup - on my Win10 system it crashes the slicer (same version as yors) with no warning or any message at all. 

What's the square box in the middle for? And this is for the F4U Corsair, yes? And not the F4 Phantom II that I worked on.

fliegermichl, 2019-11-07:
Yes it is for the F4U Corsair. The box in the middle is a wipe tower. i want to print the nose in yellow while the lower part in blue. when the filament is changing then it will print on the wipe tower to clean the nozzle from blue PLA

lukasmatena, 2019-11-07:
This slices fine in current master. I am almost sure this has the same cause as #1266 and #2258, which were solved for PrusaSlicer 2.2.0.

BirkBinnard, 2019-11-07:
Good to know Lukas - looking forward to the release of 2.2.0.

lukasmatena, 2019-11-07:
Also, it is an ecological disaster to print something like this with a wipe tower. Colorprint is a clear way to go here.

BirkBinnard, 2019-11-07:
What is colorprint?

fliegermichl, 2019-11-07:
Where can i get this Version 2.2.0? I'm pretty new to MMU with Prusa Slicer. I managed to got it running with Simplify3D but Prusa Slicer looks easier to use for me. Here is a picture from that piece sliced with simplify3d and with wipe tower. It is printing currently for 6 hours and 52 minutes.

lukasmatena, 2019-11-08:
> Where can i get this Version 2.2.0?

You can't. It has not been released yet and it is not stable. We'll release it in a few weeks (probably). It's a bit sad that you just started using PrusaSlicer and have run into a bug immediately, but hopefully there are not many more.

And like I said, you don't need to use MMU and a wipe tower to print that. There is only one toolchange at a layer boundary. PrusaSlicer can add a gcode for a manual toolchange, which will save both material and print time. See https://youtu.be/bzf20FxsN2Q?t=361

fliegermichl, 2019-11-08:
Thank you for that info. I needed a little time to find out that this manual color change is working only in MMU single mode.

closed 2019-11-08 by fliegermichl

lukasmatena, 2019-11-08:
@fliegermichl Yes, so far. It will be extended for multimaterial mode in the next release (2.2.0) or the next one.