prusa3d/PrusaSlicer #3147: Incomplete layers using three Extruders

clelong, 2019-11-07:
### Version
_Version of PrusaSlicer used goes here_
2.1 and also on 2.0
_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Windows 10 64bits
### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
A20M Geeetech
Marlin 1.1.8

### Behavior
* _Describe the problem_
Loaded three times the same stl (cube).
Slicer slices well first layer but then slices 2 cubes out of three.
Always the one with extruder T0, and alternates the one with T1 then T2.

* _Steps needed to reproduce the problem_
Loaded three times the same stl. Then sliced.

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
Slices correctly the three cubes on each layer.

* _Actual Results_
PArtly missing layers on some cubes.

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
I use 3 extruders, one is a virtual one that mix the two colors.
I loaded 3 same stl and set on different extruder on each.
When i slice, an check the G code, the first layer does its work...but then it only print 2 cubes out of 3, and alternate the extruder T0/T1, T0/T2, T0/T1, T0/T2 etc... It prints well the cube on T0 every layer, but alternates the cubes of T1 and T2 on layer out of two.
I attached the G Code...

[3 cubes 3 couleurs 20x20x10.zip](https://github.com/prusa3d/PrusaSlicer/files/3820290/3.cubes.3.couleurs.20x20x10.zip)

[image]
<img width="960" alt="other layers ko" src="https://user-images.githubusercontent.co
[20x20x10.zip](https://github.com/prusa3d/PrusaSlicer/files/3820300/20x20x10.zip)

m/14273331/68401250-b3f3ee80-0179-11ea-8b9f-433dae9dc06f.png">

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Jebtrix, 2019-11-07:
You will need to attach a zipped 3MF project file that created this gcode for people to review.

clelong, 2019-11-08:
Here are a print screen of the bed with objects, and the zip of the AMF.

[image]

[20x20x10.zip.amf.zip](https://github.com/prusa3d/PrusaSlicer/files/3823436/20x20x10.zip.amf.zip)

rtyr, 2020-04-19:
Can you try it again with 2.2.0? I am unable to reproduce the issue with your AMF.

rtyr, 2020-05-27:
No response. Closing.

closed 2020-05-27 by rtyr