prusa3d/PrusaSlicer #3148: Indexing stl files in multi colour print...How

chopcat, 2019-11-07:
### Version
2.1.0+  2109....
### Operating system type + version
MacOS 10.15.1
### 3D printer brand / version + firmware version (if known)
Prusa 3s MMSv2

Im trying to print my first multi colour print on the MMS but have not got that far aas I cant get the STL files to index when I load them into the slicer.  I have two parts for a monsy clip.  One the clip itself (colour 1) and one a 3d logo (colour 2) .  all STL fiiles are from sketchup.

When I load the logo into slicer it sticks to the z axis when i need to be able to move it up.  see pic.  is there are way to tell the slicer to keep it in the original position, whcih would be up in the air if I was not printing something to attach it to.  

Hope that makes sense

[image]

chopcat, 2019-11-07:
Sorted by a different method separating out in to objects

chopcat, 2019-11-07:
nice to know if moving of Z is possible for future use

lukasmatena, 2019-11-08:
If you load the objects at once and are using a MM printer, PrusaSlicer will ask you if you want to add them as parts of one object and in that case it will choose their relative positions based on their original coordinate system (which is probably what you want). Parts of one object can also be moved up and down, the only limitation is that one part of the object is touching the bed.

bubnikv, 2020-03-15:
answered, no reponse. closing.

closed 2020-03-15 by bubnikv