prusa3d/PrusaSlicer #3152: [Feature Request] Print area on bed as G-Code variables

mpmf, 2019-11-08:
This is a feature request.

To level the bed before printing, one generally adds a `G29` to the Start G-Code. This will probe the whole bed. However, the `G29` command supports specifying a smaller area in the form `G29 L50 F50 R150 B150`. This will probe the area from (50,50) to (150, 150). Since we generally don't care about the levelling state outside the area that will be printed, we could get more precision if we levelled the same number of points but only on the relevant area of the print bed.

What I suggest is to make available some variables so that we could add something like this to the Start G-Code:

`G29 L[print_bounding_box_left] F[print_bounding_box_front] R[print_bounding_box_right] B[pring_bounding_box_back]`

It could also be useful for file naming, storing the print size on the filename.

I believe, however, that this idea is more relevant the bigger the printer's print area is vs the actual print bounding box.

Allram, 2021-01-12:
This would be a very nice feature to have!

bubnikv, 2021-01-13:
If you read the 2.3.0 alphas change log, you will find out that the feature has been implemented.

https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-alpha1

The G-code placeholder parser supports the following new keys: "print_bed_min", "print_bed_max", "print_bed_size", which are calculated as a bounding box of an existing "bed_shape" vector of points.

The G-code placeholder parser also newly supports the following new keys: "first_layer_print_convex_hull", "first_layer_print_min", "first_layer_print_max", "first_layer_print_size", defining convex hull of the first layer. Please note that the extrusions generated by the custom G-code are not accounted for.

rtyr, 2021-02-04:
Implemented. Closing.

closed 2021-02-04 by rtyr