prusa3d/PrusaSlicer #3153: Custom marlin printer M20* codes being generated.

ril3y, 2019-11-08:
### Version
PrusaSlicer 2.1.0+win64 

### Operating system type + version
Windows

### 3D printer brand / version + firmware version (if known)
Ultimaker Original+

### Behavior
On a custom-defined marlin printer these gcodes are being inserted:
M201 X9000 Y9000 Z500 E10000 ; sets maximum accelerations, mm/sec^2
M203 X500 Y500 Z12 E120 ; sets maximum feedrates, mm/sec
M204 P1500 R1500 T1500 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X10.00 Y10.00 Z0.20 E2.50 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
Which my marlin firmware does not support.  Expected behavior is not to include these M codes
Is there a setting to remove these at slicing time?

#### Project File (.3MF) where problem occurs

M201 X9000 Y9000 Z500 E10000 ; sets maximum accelerations, mm/sec^2
M203 X500 Y500 Z12 E120 ; sets maximum feedrates, mm/sec
M204 P1500 R1500 T1500 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X10.00 Y10.00 Z0.20 E2.50 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec

ManuelMcLure, 2019-11-09:
Set your gcode flavor to RepRap in the printer settings.

lukasmatena, 2019-11-09:
@ManuelMcLure's advice is correct.

Closing, duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2567 and issues referenced from there.

closed 2019-11-09 by lukasmatena