prusa3d/PrusaSlicer #3155: Feature request: Add instances of multiple objects

tkapler, 2019-11-09:
I very often need to copy multiple small objects. Now i have to select every object and do Add instances, one by one. It would be great if i could simply use Add instances even if i have selected more objects.

And maybe one more UX improvement - long press on Add instances could show input for number of instances. And this popup could also have checkbox Auto arrange after insert (what i believe do 99 % of people in 99 % of cases anyway). And optional button - Ad as many copies as possible (or Cover the bed with copies) - what would simply Ad instance, arrange the bed, if it fits than it ads another instance ... till it does not fit and then it would delete one instance (unless there is only one of each selected objects). This would save LOTS of time for the projects with small parts you need to print many times.

JohnOCFII, 2019-11-09:
This doesn't help with selecting multiple different objects, but there is the multiplier option there today as shown in this screen shot.  
[image]

neophyl, 2019-11-09:
Why do'nt you select the multiple objects you want, then cntrl-c to copy them, then simply cntrl-v to paste a copy of the selected objects ?.  You can keep pasting them over and over and hit arrange all, when ever you want to see them laid out.  If there's still space just cntrl-v a few more times and arrange.  Its a very quick way to fill up the bed with multiple small objects.

tkapler, 2019-11-09:
Of course i know both of this, but both hints are useless
1. number of copies - yes, that is fine, but you will not fill the bed, so if you do not know if you need 30 or 50 objects, you end up with tons of clicking. And it does not work for multiple objects.
2. copy and paste is NOT creating another instance! It is creating new object. That is very big difference, technicaly new instance is few bytes (position of the second copy) comparing to completely new STL content. practicaly - changes on one copy does not propagate to the others, but it works for instances.

lukasmatena, 2019-11-10:
@tkapler Excuse me, but is the 'fill bed' part of the request an exact duplicate of YOUR OWN issue #1350? Kindly discuss it there and don't clutter the issue tracker more than it already is. Thanks.

The idea that Add/Remove instance could work with multiple-object selection is something that might be possible, although I'm not sure how many use cases it has. I'm also not sure how easy that would be implement and what implications it has @YuSanka .

tkapler, 2019-11-11:
@lukasmatena you are right, i forgot that in my #1350 issue last year i mentioned also multiplication of multiple objects. But those are two tasks
- fill the bed (even with one object) (#1350)
- copy instances of multiple object (this task)

But yes, here in this task i also cover how it could work for both tasks.

tkapler, 2019-11-11:
@lukasmatena and about the need of such function - i am a member of Prusalab, where is about 20 printers printing whatever members needs. And almost always there is one or more printers printing something, where author wants to print multiple copies of the same thing and want to cover the bed with them, quite often with multiparts (but they often use one printer for one part and second for the other). So I think this is quite common scenario.

closed 2026-09-29 by prusaslicer-bot[bot]