prusa3d/PrusaSlicer #3177: Prusa Slicer Crashes

ggcarmichael, 2019-11-12:
### Version
Version: 2.1.0+win64
Build: PrusaSlicer-2.1.0+win64-201909160915

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 18362), 64-bit edition
Total RAM size [MB]: 17,058MB

### 3D printer brand / version + firmware version (if known)
 Prusa i3 MK3S
FW3.8.1

### Behavior
Application shuts down unexpectedly after opening.  Doesn't seem to be related to performing a particular action.  I have uninstalled and reinstalled 3 times.  Issue continues.  Cannot use this Slicr version

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

tamasmeszaros, 2019-11-12:
Have you tried removing all the profiles and cache data? Its in `Users\<userdir>\AppData\Roaming\PrusaSlicer`

lukasmatena, 2019-11-15:
@ggcarmichael Let me guess.. nVidia graphics card?

Could you try to run PrusaSlicer from command line as `prusa-slicer-console.exe --sw-renderer --loglevel=9` and tell us what happened?

StageMonkey, 2019-12-17:
Has this worked for anyone?  Has anyone found another solution?  I'm unable to get it to work and am still dealing with random crashing of prusaslicer.  It is completely unusable for me right now.  It won't stay running long enough to do anything.

lukasmatena, 2019-12-17:
There is a hint in my last post about what to try to help us debug the issue. Could you provide the info that @ggcarmichael hasn't? Thanks.

StageMonkey, 2019-12-17:
I tried the command line, but I get an error - 'prusa-slicer-console.exe' is not recognized as an internal or external command, operable program or batch file.'  It's likely I'm doing something wrong, but I don't know what it would be.  I did copy the files in the cache  folder to a backup and tried loading PrusaSlicer, and it's still working for now.  I'll report back if it continues to work or not. 

For the record, I'm on a Win10/64bit machine with 12G of ram with Intel HD Graphics 520.  I have a Prusa MK2S (upgrades to 2.5).  Been using PrusaSlicer/Slic3r for a couple years on this machine.  The crashing started happening an update or two ago

lukasmatena, 2019-12-18:
@StageMonkey The error with unrecognized command suggests that you didn't `cd` into the correct directory. The `--sw-renderer` parameter would turn off hardware acceleration, so the rendering would be slow, but bypass issues related to that. And there were some issues with Intel integrated graphics cards in the past.

Are you saying that moving the config folder and running a wizard from scratch on the next run helped? If so, the issue would not be caused by the graphics, but it would indicate corrupted config. If you still have the malfunctioning directory somewhere, we would like to see it. There was an issue with corrupted config on Win already (#2466), this may or may not be related. If it appears again, try the `--loglevel 9` parameter to get a log, it might give us more.

StageMonkey, 2019-12-18:
I moved whatever files were in AppData/roaming/PrusaSlicer/cache into a newly renamed folder ultimately emptying the existing cache folder. After doing so, I was able to open PrusaSlicer and it didn’t crash on me. Traditionally it would crash at random. Sometimes right after starting it. Other times right before I sliced a model. So far it is continuing to open and function properly. 
I did cd to what I believe to be the proper directory when I tried the command prompt trick.  Can you tell me which directory I should be in just in case I was trying the wrong one?    	    	
Thanks!      From: lukasmatena <notifications@github.com>
Sent: Tuesday, December 17, 2019 11:01 PM
To: prusa3d/PrusaSlicer
Cc: StageMonkey; Mention
Subject: Re: [prusa3d/PrusaSlicer] Prusa Slicer Crashes (#3177) 

@StageMonkey The error with unrecognized command suggests that you didn't cd into the correct directory. The --sw-renderer parameter would turn off hardware acceleration, so the rendering would be slow, but bypass issues related to that. And there were some issues with Intel integrated graphics cards in the past.

Are you saying that moving the config folder and running a wizard from scratch on the next run helped? If so, the issue would not be caused by the graphics, but it would indicate corrupted config. If you still have the malfunctioning directory somewhere, we would like to see it. There was an issue with corrupted config on Win already (#2466), this may or may not be related. If it appears again, try the --loglevel 9 parameter to get a log, it might give us more.

lukasmatena, 2019-12-18:
@StageMonkey Thanks. Could you zip the cache directory and attach it here, so we see if and how it is corrupted? Actually, if you can share your profiles, zipping the whole PrusaSlicer config directory would be even better.

As for the directory, it is the directory where that executable is downloaded/installed. When you type `dir`, it should list the `prusa-slicer-console.exe` (among others). But it is probably no longer needed. (Unless you want to help us debug it more - in that case, get it back to the state when it crashes and then get the log so we see at which point it crashes. But providing your config might allow us to reproduce it anyway.

lukasmatena, 2019-12-19:
@StageMonkey I quickly looked at that config. Files in cache and cache BAK are identical and seem not to be corrupted. I doubt the problem is config related. Try disabling the hardware acceleration to see if it goes away. Also, a log might reveal more about the point where it crashes. Thanks.

StageMonkey, 2019-12-19:
So I ran the command, and this was the what came up.  Let me know if this does you any good.  It did open PrusaSlicer, but it didn't stay open.  Was that supposed to happen?

`Loading MESA OpenGL library: C:\Program Files\Prusa3D\PrusaSlicer\mesa\opengl32.dll
MESA OpenGL library was loaded sucessfully
[2019-12-19 08:42:38.032578] [0x00021ef0] [info]    Checking if indices need to be installed from resources...
[2019-12-19 08:42:38.167216] [0x00021ef0] [trace]   System language detected (user locales and such): en_US
[2019-12-19 08:42:38.172200] [0x00021ef0] [trace]   Best translation language detected (may be different from user locales): en_GB
[2019-12-19 08:42:38.182186] [0x00021ef0] [trace]   Switching wxLocales to en_GB
[2019-12-19 08:42:43.445815] [0x00021ef0] [info]    Undo / Redo snapshot taken: New Project, Undo / Redo stack memory: 0MB WorkingSet: 35MB; PrivateBytes: 16MB; Pagefile(peak): 16MB(17MB)
[2019-12-19 08:43:01.496212] [0x00021ef0] [debug]   TriangleMesh::repair() started
[2019-12-19 08:43:01.529144] [0x00021ef0] [debug]   TriangleMesh::repair() finished
[2019-12-19 08:43:01.545084] [0x00021ef0] [trace]   TriangleMeshSlicer::require_shared_vertices - start
[2019-12-19 08:43:01.548075] [0x00021ef0] [trace]   TriangleMeshSlicer::require_shared_vertices - stl_generate_shared_vertices
[2019-12-19 08:43:01.559052] [0x00021ef0] [trace]   TriangleMeshSlicer::require_shared_vertices - end
[2019-12-19 08:43:01.588844] [0x00021ef0] [debug]   TriangleMesh::repair() started
[2019-12-19 08:43:01.599829] [0x00021ef0] [debug]   TriangleMesh::repair() finished
[2019-12-19 08:43:01.610647] [0x00021ef0] [trace]   TriangleMeshSlicer::require_shared_vertices - start
[2019-12-19 08:43:01.614635] [0x00021ef0] [trace]   TriangleMeshSlicer::require_shared_vertices - stl_generate_shared_vertices
[2019-12-19 08:43:01.634582] [0x00021ef0] [trace]   TriangleMeshSlicer::require_shared_vertices - end
[2019-12-19 08:43:06.254219] [0x00021ef0] [info]    Checking for cached configuration updates...
[2019-12-19 08:43:06.323054] [0x00021ef0] [debug]   Vendor: Prusa Research, version installed: 1.0.5, version cached: 1.0.7
[2019-12-19 08:43:06.452359] [0x00021ef0] [info]    Downloaded index for `PrusaResearch` is the same as installed one, not offering an update.
[2019-12-19 08:43:06.489398] [0x00021ef0] [info]    No configuration updates available.
[2019-12-19 08:43:06.742193] [0x00021b98] [info]    Downloading PrusaSlicer online version from: `http://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version`
*   Trying 217.16.180.252...
* TCP_NODELAY set
* Connected to files.prusa3d.com (217.16.180.252) port 80 (#0)
> GET /wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version HTTP/1.1
Host: files.prusa3d.com
User-Agent: PrusaSlicer/2.1.1+win64
Accept: */*

< HTTP/1.1 200 OK
< Server: nginx
< Date: Thu, 19 Dec 2019 16:43:04 GMT
< Content-Length: 6
< Connection: keep-alive
< X-Accel-Version: 0.01
< Last-Modified: Thu, 19 Dec 2019 16:03:43 GMT
< ETag: "6-59a10b49645c1"
< Accept-Ranges: bytes
< X-Powered-By: PleskLin
<
* Connection #0 to host files.prusa3d.com left intact
[2019-12-19 08:43:07.294767] [0x00021b98] [info]    Got PrusaSlicer online version: `2.1.1`. Sending to GUI thread...
[2019-12-19 08:43:07.316716] [0x00021b98] [info]    Syncing configuration cache
[2019-12-19 08:43:07.334666] [0x00021b98] [info]    Downloading index for vendor: Prusa Research
[2019-12-19 08:43:07.344633] [0x00021b98] [info]    Get: `http://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaResearch//index.idx`
        -> `C:\Users\jjbgo\AppData\Roaming\PrusaSlicer\cache\PrusaResearch.idx-update`
        via tmp path `C:\Users\jjbgo\AppData\Roaming\PrusaSlicer\cache\PrusaResearch.idx-update.141860.download`
*   Trying 217.16.180.252...
* TCP_NODELAY set
* Connected to files.prusa3d.com (217.16.180.252) port 80 (#0)
> GET /wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaResearch//index.idx HTTP/1.1
Host: files.prusa3d.com
User-Agent: PrusaSlicer/2.1.1+win64
Accept: */*

< HTTP/1.1 200 OK
< Server: nginx
< Date: Thu, 19 Dec 2019 16:43:05 GMT
< Content-Type: application/octet-stream
< Content-Length: 5824
< Last-Modified: Thu, 19 Dec 2019 16:03:43 GMT
< Connection: keep-alive
< ETag: "5dfb9f5f-16c0"
< X-Powered-By: PleskLin
< Accept-Ranges: bytes
<
* Connection #0 to host files.prusa3d.com left intact
[2019-12-19 08:43:07.823858] [0x00021b98] [debug]   Got index for vendor: Prusa Research: current version: 1.0.5, recommended version: 1.0.7
[2019-12-19 08:43:07.860330] [0x00021b98] [info]    Downloading new bundle for vendor: Prusa Research
[2019-12-19 08:43:07.890252] [0x00021b98] [info]    Get: `http://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaResearch//1.0.7.ini`
        -> `C:\Users\jjbgo\AppData\Roaming\PrusaSlicer\cache\PrusaResearch.ini`
        via tmp path `C:\Users\jjbgo\AppData\Roaming\PrusaSlicer\cache\PrusaResearch.ini.141860.download`
*   Trying 217.16.180.252...
* TCP_NODELAY set
* Connected to files.prusa3d.com (217.16.180.252) port 80 (#0)
> GET /wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaResearch//1.0.7.ini HTTP/1.1
Host: files.prusa3d.com
User-Agent: PrusaSlicer/2.1.1+win64
Accept: */*

< HTTP/1.1 200 OK
< Server: nginx
< Date: Thu, 19 Dec 2019 16:43:05 GMT
< Content-Type: application/octet-stream
< Content-Length: 159132
< Last-Modified: Thu, 19 Dec 2019 16:03:43 GMT
< Connection: keep-alive
< ETag: "5dfb9f5f-26d9c"
< X-Powered-By: PleskLin
< Accept-Ranges: bytes
<
* Connection #0 to host files.prusa3d.com left intact`

StageMonkey, 2019-12-19:
Tried again without the log, and the software crashed before I could even load a model.  Below are the results from that:

`C:\Program Files\Prusa3D\PrusaSlicer>prusa-slicer-console.exe --sw-renderer
Loading MESA OpenGL library: C:\Program Files\Prusa3D\PrusaSlicer\mesa\opengl32.dll
MESA OpenGL library was loaded sucessfully
Qt: Untested Windows version 10.0 detected!
Qt: Untested Windows version 10.0 detected!
Qt: Untested Windows version 10.0 detected!
Qt: Untested Windows version 10.0 detected!
Qt: Untested Windows version 10.0 detected!
Qt: Untested Windows version 10.0 detected!`

StageMonkey, 2019-12-28:
Any thoughts?  PrusaSlicer is completely unusable for me now, and I would love to get it back.

StageMonkey, 2019-12-28:
I've completely removed all prusa software and reinstalled.  I ran the sw-renderer in command line with the new install, and it still crashed. But this time it waited until I was able to load and slice an object.  The results can be found here.  Warning, it was a large object that was sliced, so this file is a bit long.  The results were saved to google drive -   [https://docs.google.com/document/d/1QKw56WGcadmHTp0wPp6GZT96NW7g-DjzjYjXUXcNHg4/edit?usp=sharing](url)
``

Hostile17, 2020-02-08:
im getting crashes too. i have tried everything mentioned here, and more. this latest version just is not stable. sometime it will work fine, other times crashes when slicing. or it will slice, then crash when i export...  hopefully the 2.2 version will be more stable.

bubnikv, 2020-02-09:
@StageMonkey 
> C:\Program Files\Prusa3D\PrusaSlicer>prusa-slicer-console.exe --sw-renderer Loading MESA OpenGL library: C:\Program Files\Prusa3D\PrusaSlicer\mesa\opengl32.dll MESA OpenGL library was loaded [...]

This makes me nervous. Who is writing out Qt: Untested Windows version 10 detected! ?
Is it some 3rd party application that plugs into other applications? What did you install on your computer? It may very well be that PrusaSlicer is not guilty.

@StageMonkey @Hostile17 Which versions of PrusaSlicer run correctly on your computer and wich do not? Did you try the latest alpha? https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.2.0-alpha4

midnightsmith89, 2022-01-13:
Any follow up to this? I'm having similar issues. If I let the program sit for 5 minutes, and go to load an stl, it crashes, but ONLY when I got to load an stl from the built in add button. I can drag and drop from explorer fine.

lukasmatena, 2023-02-06:
Could you guys please retest this with [2.6.0-alpha3](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha3)? Thanks.

MrDach, 2023-07-31:
I'm also seeing this error.
I just installed 2.6.0 because I have a Prusa XL incoming soon (hopefully) and wanted to switch to PrusaSlicer instead of Cura beforehand.

Dell latitude 5401 running Windows 11 and a Geforce MX150.

Whenever the program crashes I'm unable to start it again and have to restart the laptop to fix it.

lukasmatena, 2023-07-31:
@MrDach Thanks for the report. We need to know more to be able to debug it, the issue may be something else than what the others are reporting.

1. I understand that you can run the application, but after it crashes, you can no longer run it until you reboot the system. Is that correct?
2. When does the application crash? What do you do before it happens?
3. Could you run the application from the console as `prusa-slicer-console --sw-renderer` and check if the problem is the same?
4. Could you run the application from the console as `prusa-slicer-console.exe --loglevel=5` and see if there is anything interesting in the log during the first run (that crashes) or the others (those that don't start)?
5. Are you able to provide some debugging output? If you enable crash dump generation on your system according to https://helgeklein.com/blog/creating-an-application-crash-dump/, let the application crash and send us the generated dump, we might be able to see what is going on.

Thanks.

lukasmatena, 2024-02-18:
No response, most likely outdated. Closing.

closed 2024-02-18 by lukasmatena