prusa3d/PrusaSlicer #3179: Crash of slicer after "Gyroid improvements"

rmoravcik, 2019-11-12:
### Version
Self compiled version from git (f7e27eb21d745eaad14b31cd777700a9f6a6c6d1)

### Operating system type + version
Ubuntu 19.10

### Behavior
PrusaSlicer is crashing when trying to slice this object https://www.thingiverse.com/download:2038282

Here is also output from gdb:
```
morgan@morgan:~/Projekty/PrusaSlicer/PrusaSlicer/_build$ gdb ./src/prusa-slicer 
GNU gdb (Ubuntu 8.3.50.20190910-0ubuntu1) 8.3.50.20190910-git
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./src/prusa-slicer...
(No debugging symbols found in ./src/prusa-slicer)
(gdb) run
Starting program: /home/morgan/Projekty/PrusaSlicer/PrusaSlicer/_build/src/prusa-slicer 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/x86_64-linux-gnu/libthread_db.so.1".
Gtk-Message: 13:33:15.911: Failed to load module "canberra-gtk-module"
[Detaching after vfork from child process 58319]
[Detaching after vfork from child process 58321]
[Detaching after vfork from child process 58323]
[Detaching after vfork from child process 58325]
[Detaching after vfork from child process 58327]
[Detaching after vfork from child process 58329]
[Detaching after vfork from child process 58331]
[Detaching after vfork from child process 58333]
[New Thread 0x7ffff0b23700 (LWP 58335)]
[New Thread 0x7ffff0303700 (LWP 58336)]
[New Thread 0x7fffefb02700 (LWP 58337)]
[New Thread 0x7fffef301700 (LWP 58338)]
[Thread 0x7fffef301700 (LWP 58338) exited]

(prusa-slicer:58315): Gtk-CRITICAL **: 13:33:20.027: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed

(prusa-slicer:58315): Gtk-CRITICAL **: 13:33:20.886: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed

(prusa-slicer:58315): Gtk-CRITICAL **: 13:33:20.898: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed

(prusa-slicer:58315): Gtk-CRITICAL **: 13:33:20.898: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed

(prusa-slicer:58315): Gtk-CRITICAL **: 13:33:20.898: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed

(prusa-slicer:58315): Gtk-CRITICAL **: 13:33:20.898: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed

(prusa-slicer:58315): Gtk-CRITICAL **: 13:33:20.898: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed

(prusa-slicer:58315): Gtk-CRITICAL **: 13:33:20.898: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed

(prusa-slicer:58315): Gtk-CRITICAL **: 13:33:20.898: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed

(prusa-slicer:58315): Gtk-CRITICAL **: 13:33:20.898: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed

(prusa-slicer:58315): Gtk-CRITICAL **: 13:33:20.898: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed

(prusa-slicer:58315): Gtk-CRITICAL **: 13:33:20.898: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed

(prusa-slicer:58315): Gtk-CRITICAL **: 13:33:20.898: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed

(prusa-slicer:58315): Gtk-CRITICAL **: 13:33:20.898: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
[New Thread 0x7fffef301700 (LWP 58343)]
[New Thread 0x7fffecd01700 (LWP 58344)]
[Thread 0x7fffecd01700 (LWP 58344) exited]
[Thread 0x7fffef301700 (LWP 58343) exited]
[New Thread 0x7fffecd01700 (LWP 58345)]
[Thread 0x7fffefb02700 (LWP 58337) exited]

(prusa-slicer:58315): Gtk-CRITICAL **: 13:33:25.238: IA__gtk_tree_row_reference_new_proxy: assertion 'path->depth > 0' failed

(prusa-slicer:58315): Gtk-CRITICAL **: 13:33:25.239: IA__gtk_tree_row_reference_new_proxy: assertion 'path->depth > 0' failed

(prusa-slicer:58315): Gtk-CRITICAL **: 13:33:25.244: IA__gtk_tree_row_reference_new_proxy: assertion 'path->depth > 0' failed

(prusa-slicer:58315): Gtk-CRITICAL **: 13:33:25.257: IA__gtk_tree_row_reference_new_proxy: assertion 'path->depth > 0' failed

(prusa-slicer:58315): Gtk-CRITICAL **: 13:33:25.257: IA__gtk_tree_row_reference_new_proxy: assertion 'path->depth > 0' failed

(prusa-slicer:58315): Gtk-CRITICAL **: 13:33:25.258: IA__gtk_tree_row_reference_new_proxy: assertion 'path->depth > 0' failed

(prusa-slicer:58315): Gtk-CRITICAL **: 13:33:25.258: IA__gtk_tree_row_reference_new_proxy: assertion 'path->depth > 0' failed
[New Thread 0x7fffefb02700 (LWP 58348)]

Thread 9 "prusa-slicer" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffefb02700 (LWP 58348)]
0x0000555555bd1f30 in Slic3r::mark_boundary_segments_touching_infill(std::vector<std::vector<Slic3r::Point, std::allocator<Slic3r::Point> >, std::allocator<std::vector<Slic3r::Point, std::allocator<Slic3r::Point> > > > const&, std::vector<std::vector<Slic3r::ContourPointData, std::allocator<Slic3r::ContourPointData> >, std::allocator<std::vector<Slic3r::ContourPointData, std::allocator<Slic3r::ContourPointData> > > >&, Slic3r::BoundingBox const&, std::vector<Slic3r::Polyline, std::allocator<Slic3r::Polyline> > const&, double, double)::Visitor::operator()(int, int) ()
(gdb) bt
#0  0x0000555555bd1f30 in Slic3r::mark_boundary_segments_touching_infill(std::vector<std::vector<Slic3r::Point, std::allocator<Slic3r::Point> >, std::allocator<std::vector<Slic3r::Point, std::allocator<Slic3r::Point> > > > const&, std::vector<std::vector<Slic3r::ContourPointData, std::allocator<Slic3r::ContourPointData> >, std::allocator<std::vector<Slic3r::ContourPointData, std::allocator<Slic3r::ContourPointData> > > >&, Slic3r::BoundingBox const&, std::vector<Slic3r::Polyline, std::allocator<Slic3r::Polyline> > const&, double, double)::Visitor::operator()(int, int) ()
#1  0x0000555555bd39f0 in Slic3r::mark_boundary_segments_touching_infill(std::vector<std::vector<Slic3r::Point, std::allocator<Slic3r::Point> >, std::allocator<std::vector<Slic3r::Point, std::allocator<Slic3r::Point> > > > const&, std::vector<std::vector<Slic3r::ContourPointData, std::allocator<Slic3r::ContourPointData> >, std::allocator<std::vector<Slic3r::ContourPointData, std::allocator<Slic3r::ContourPointData> > > >&, Slic3r::BoundingBox const&, std::vector<Slic3r::Polyline, std::allocator<Slic3r::Polyline> > const&, double, double) ()
#2  0x0000555555bd4ac4 in Slic3r::Fill::connect_infill(std::vector<Slic3r::Polyline, std::allocator<Slic3r::Polyline> >&&, Slic3r::ExPolygon const&, std::vector<Slic3r::Polyline, std::allocator<Slic3r::Polyline> >&, Slic3r::FillParams const&) ()
#3  0x0000555555bdba92 in Slic3r::FillGyroid::_fill_surface_single(Slic3r::FillParams const&, unsigned int, std::pair<float, Slic3r::Point> const&, Slic3r::ExPolygon&, std::vector<Slic3r::Polyline, std::allocator<Slic3r::Polyline> >&) ()
#4  0x0000555555bd1d88 in Slic3r::Fill::fill_surface(Slic3r::Surface const*, Slic3r::FillParams const&) ()
#5  0x0000555555bce585 in Slic3r::Layer::make_fills() ()
#6  0x0000555555ab046e in tbb::interface9::internal::start_for<tbb::blocked_range<unsigned long>, Slic3r::PrintObject::infill()::{lambda(tbb::blocked_range<unsigned long> const&)#1}, tbb::auto_partitioner const>::execute() ()
#7  0x00007ffff6b4f09d in ?? () from /usr/lib/x86_64-linux-gnu/libtbb.so.2
#8  0x00007ffff6b4c468 in ?? () from /usr/lib/x86_64-linux-gnu/libtbb.so.2
#9  0x0000555555aa9989 in Slic3r::PrintObject::infill() ()
#10 0x0000555555a52d4c in Slic3r::Print::process() ()
#11 0x0000555555ec8b84 in Slic3r::BackgroundSlicingProcess::process_fff() ()
#12 0x0000555555ec95d8 in Slic3r::BackgroundSlicingProcess::thread_proc() ()
#13 0x0000555555ec978a in Slic3r::BackgroundSlicingProcess::thread_proc_safe() ()
#14 0x00007ffff6d961b5 in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.67.0
#15 0x00007ffff69a1669 in start_thread (arg=<optimized out>) at pthread_create.c:479
#16 0x00007ffff68c9323 in clone () from /usr/lib/x86_64-linux-gnu/libc.so.6
(gdb) quit
A debugging session is active.

	Inferior 1 [process 58315] will be killed.

Quit anyway? (y or n) y
morgan@morgan:~/Projekty/PrusaSlicer/PrusaSlicer/_build$
```

lukasmatena, 2019-11-12:
Thanks, I can confirm the crash. Also, binary compiled with ASAN seems to crash with almost any model that is high enough. I enclose a 3MF which crashes. Changing object height to 7mm makes the crash disappear.

[gyroid_crash_3179.zip](https://github.com/prusa3d/PrusaSlicer/files/3836305/gyroid_crash_3179.zip)

rmoravcik, 2019-11-15:
I just tried commit https://github.com/prusa3d/PrusaSlicer/commit/dd59945098fbf55eb45e30d67471f8f265895a65 and it seems it fixed my problems. I'm not able to reproduce crash anymore.

closed 2019-11-15 by rmoravcik