prusa3d/PrusaSlicer #3182: slicer GUI build not slicing on Mac OS

Anmol6, 2019-11-13:
### Version
version_2.1.0-485-g050a9adf3

### Operating system type + version
OS: Mac OS Mojave 10.14.6

### Behavior
So I built the slicer for mac os following the instructions provided. When I click on the `build/src/prusa-slicer` file, the gui opens just fine. However, when I load any stl and try to slice it, I get a segmentation fault. Here's what the corresponding terminal shows 
```
objc[3080]: Class FIFinderSyncExtensionHost is implemented in both /System/Library/PrivateFrameworks/FinderKit.framework/Versions/A/FinderKit (0x7fffad89d3d8) and /System/Library/PrivateFrameworks/FileProvider.framework/OverrideBundles/FinderSyncCollaborationFileProviderOverride.bundle/Contents/MacOS/FinderSyncCollaborationFileProviderOverride (0x1196ddf50). One of the two will be used. Which one is undefined.
Segmentation fault: 11
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.
```

Would appreciate any help! 

Is this a new feature request? No

lukasmatena, 2019-11-13:
Could you try to run the executable with parameter `--loglevel 9` to get more extensive log? Thanks.

norre-daroy, 2019-11-14:
@Anmol6  I have the same issue . Please let me know if you have made some progress. Thanks

lukasmatena, 2019-11-14:
@Anmol6  @norre-daroy  Could either of you provide that log? Also try to go to Print Settings->Infill and choose different pattern than Gyroid (which is the default in our profiles), there was recently an issue with it.

lukasmatena, 2019-11-15:
@Anmol6 @norre-daroy Could you please retest with latest master 32d794756ee2e7d43bafd07180e3e6f05e000265 ?

A serious bug has been fixed with dd59945098fbf55eb45e30d67471f8f265895a65, there is a chance it was a cause of your crashes as well.

norre-daroy, 2019-11-15:
Hello @lukasmatena . I pulled the latest master and I got these errors building it
[image]

norre-daroy, 2019-11-15:
@lukasmatena I tried to build it again with the latest master 32d7947 changes and I got these errors
[image]

This is the git log
[image]

lukasmatena, 2019-11-15:
@norre-daroy What you posted is just a report that an error appeared at some point before. You must upscroll to see the actual error message. The "expected unqualified-id" sounds serious, but I cannot see where exactly it comes from (could be the Config.cpp or anything included by it).

The commit builds fine on my Ubuntu 16.04, but the master branch is under rapid development and it is possible that some recent change broke build on OSX (or at least under some circumstances). I need to see the whole `make` output to be able to guess what may be wrong and if it is at your end or in the codebase.

norre-daroy, 2019-11-16:
@lukasmatena Sorry, I upscroll and pasted the message 

Scanning dependencies of target imgui
Scanning dependencies of target glew
Scanning dependencies of target encoding-check
Scanning dependencies of target expat
[  1%] Building CXX object src/build-utils/CMakeFiles/encoding-check.dir/encoding-check.cpp.o
[  1%] Building C object CMakeFiles/glew.dir/src/glew/src/glew.c.o
[  2%] Building C object CMakeFiles/expat.dir/src/expat/xmlparse.c.o
[  3%] Building CXX object src/imgui/CMakeFiles/imgui.dir/imgui.cpp.o
[  3%] Linking CXX executable encoding-check
[  3%] Built target encoding-check
[  3%] Building C object CMakeFiles/expat.dir/src/expat/xmlrole.c.o
Scanning dependencies of target admesh
[  3%] Building CXX object src/admesh/CMakeFiles/admesh.dir/connect.cpp.o
[  3%] Building C object CMakeFiles/expat.dir/src/expat/xmltok.c.o
[  4%] Linking C static library libexpat.a
[  4%] Built target expat
Scanning dependencies of target avrdude-conf-gen
[  4%] Building CXX object src/avrdude/CMakeFiles/avrdude-conf-gen.dir/conf-generate.cpp.o
/Users/ninja.n/WebDev/ORDSolutions/PrusaSlicer/src/admesh/connect.cpp:219:16: warning: 
      unused variable 'temp' [-Wunused-variable]
                                        HashEdge *temp = link->next;
                                                  ^
[  5%] Linking CXX executable avrdude-conf-gen
[  5%] Built target avrdude-conf-gen
[  5%] Building CXX object src/imgui/CMakeFiles/imgui.dir/imgui_demo.cpp.o
Scanning dependencies of target nowide
[  5%] Building CXX object src/boost/CMakeFiles/nowide.dir/nowide/iostream.cpp.o
1 warning generated.
[  5%] Building CXX object src/admesh/CMakeFiles/admesh.dir/normals.cpp.o
[  5%] Linking CXX static library libnowide.a
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: libnowide.a(iostream.cpp.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: libnowide.a(iostream.cpp.o) has no symbols
warning: /Library/Developer/CommandLineTools/usr/bin/ranlib: warning for library: libnowide.a the table of contents is empty (no object file members in the library define global symbols)
[  5%] Built target nowide
[  5%] Building CXX object src/imgui/CMakeFiles/imgui.dir/imgui_draw.cpp.o
Scanning dependencies of target clipper
[  6%] Building CXX object src/clipper/CMakeFiles/clipper.dir/clipper.cpp.o
/Users/ninja.n/WebDev/ORDSolutions/PrusaSlicer/src/admesh/normals.cpp:195:21: warning: 
      unused variable 'temp' [-Wunused-variable]
                stl_normal *temp = head->next;  // Delete this facet fro...
                            ^
1 warning generated.
[  7%] Building CXX object src/admesh/CMakeFiles/admesh.dir/shared.cpp.o
[  8%] Building CXX object src/imgui/CMakeFiles/imgui.dir/imgui_widgets.cpp.o
[  8%] Building CXX object src/admesh/CMakeFiles/admesh.dir/stl_io.cpp.o
[  8%] Building CXX object src/clipper/CMakeFiles/clipper.dir/clipper_z.cpp.o
[  8%] Building CXX object src/admesh/CMakeFiles/admesh.dir/stlinit.cpp.o
[  8%] Linking CXX static library libimgui.a
[  8%] Built target imgui
[  9%] Building CXX object src/admesh/CMakeFiles/admesh.dir/util.cpp.o
In file included from /Users/ninja.n/WebDev/ORDSolutions/PrusaSlicer/src/admesh/stlinit.cpp:31:
In file included from /Users/ninja.n/WebDev/ORDSolutions/PrusaSlicer/deps/build/destdir/usr/local/include/boost/detail/endian.hpp:9:
/Users/ninja.n/WebDev/ORDSolutions/PrusaSlicer/deps/build/destdir/usr/local/include/boost/predef/detail/endian_compat.h:11:9: warning: 
      The use of BOOST_*_ENDIAN and BOOST_BYTE_ORDER is deprecated. Please
      include <boost/predef/other/endian.h> and use BOOST_ENDIAN_*_BYTE instead
      [-W#pragma-messages]
#pragma message("The use of BOOST_*_ENDIAN and BOOST_BYTE_ORDER is depre...
        ^
1 warning generated.
Scanning dependencies of target miniz_static
[  9%] Building C object src/miniz/CMakeFiles/miniz_static.dir/miniz.c.o
[  9%] Linking CXX static library libclipper.a
[  9%] Built target clipper
Scanning dependencies of target glu-libtess
[  9%] Building C object src/glu-libtess/CMakeFiles/glu-libtess.dir/src/dict.c.o
[  9%] Linking CXX static library libadmesh.a
[  9%] Building C object src/glu-libtess/CMakeFiles/glu-libtess.dir/src/geom.c.o
[  9%] Built target admesh
[... 691 lines cut ...]
    template <class _Tp, class _Alloc> friend class _LIBCPP_TEMPLATE_VIS vector;
                                                                         ^
/Library/Developer/CommandLineTools/usr/include/c++/v1/memory:1880:64: note: 
      qualify call to silence this warning
    _LIBCPP_INLINE_VISIBILITY void destroy(pointer __p) {__p->~_Tp();}
                                                               ^
In file included from /Users/ninja.n/WebDev/ORDSolutions/PrusaSlicer/src/libslic3r/Config.cpp:2:
/Users/ninja.n/WebDev/ORDSolutions/PrusaSlicer/src/libslic3r/Utils.hpp:197:17: error: 
      expected a qualified name after 'typename'
inline typename const CONTAINER_TYPE::value_type& prev_value_modulo(type...
                ^
/Users/ninja.n/WebDev/ORDSolutions/PrusaSlicer/src/libslic3r/Utils.hpp:197:17: error: 
      expected unqualified-id
2 errors generated.
make[2]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/Config.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
4 warnings generated.
make[1]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
1 warning generated.
[ 36%] Linking CXX executable libnest2d_tests
[ 36%] Built target libnest2d_tests
[ 37%] Linking C static library libglew.a
[ 37%] Built target glew
make: *** [all] Error 2

lukasmatena, 2019-11-16:
@norre-daroy Thanks, I tried to fix it with https://github.com/prusa3d/PrusaSlicer/commit/bbb519ad0eceba5a2a6812b853fa2192bd8c9594. Please, try that.

norre-daroy, 2019-11-18:
@lukasmatena The fix bbb519a worked. But the slicer GUI still not slicing. Below is the log

 Norres-MacBook-Air:build ninja.n$ /Users/ninja.n/WebDev/ORDSolutions/PrusaSlicer/build/src/prusa-slicer --loglevel 9
[2019-11-18 13:53:28.510074] [0x000000010eeee5c0] [info]    Checking if indices need to be installed from resources...
[2019-11-18 13:53:28.533695] [0x000000010eeee5c0] [trace]   System language detected (user locales and such): en_CA
[2019-11-18 13:53:28.545328] [0x000000010eeee5c0] [trace]   Best translation language detected (may be different from user locales): en_GB
[2019-11-18 13:53:28.545386] [0x000000010eeee5c0] [trace]   Switching wxLocales to en_GB
[2019-11-18 13:53:28.937743] [0x000000010eeee5c0] [debug]   GLCanvas3D: Use Retina OpenGL: true
[2019-11-18 13:53:28.937774] [0x000000010eeee5c0] [debug]   GLCanvas3D: Use Retina OpenGL: true
[2019-11-18 13:53:28.942486] [0x000000010eeee5c0] [info]    Undo / Redo snapshot taken: New Project, Undo / Redo stack memory: 0MB Resident memory: 37MB; Peak memory usage: 37MB
[2019-11-18 13:53:30.335665] [0x000000010eeee5c0] [debug]   GLCanvas3D: Use Retina OpenGL: true
[2019-11-18 13:53:30.335705] [0x000000010eeee5c0] [debug]   GLCanvas3D: Use Retina OpenGL: true
[2019-11-18 13:53:31.367768] [0x000000010eeee5c0] [debug]   TriangleMesh::repair() started
[2019-11-18 13:53:31.371182] [0x000000010eeee5c0] [debug]   TriangleMesh::repair() finished
[2019-11-18 13:53:31.371215] [0x000000010eeee5c0] [trace]   TriangleMeshSlicer::require_shared_vertices - start
[2019-11-18 13:53:31.371224] [0x000000010eeee5c0] [trace]   TriangleMeshSlicer::require_shared_vertices - stl_generate_shared_vertices
[2019-11-18 13:53:31.371560] [0x000000010eeee5c0] [trace]   TriangleMeshSlicer::require_shared_vertices - end
[2019-11-18 13:53:31.379499] [0x000000010eeee5c0] [debug]   TriangleMesh::repair() started
[2019-11-18 13:53:31.380024] [0x000000010eeee5c0] [debug]   TriangleMesh::repair() finished
[2019-11-18 13:53:31.380041] [0x000000010eeee5c0] [trace]   TriangleMeshSlicer::require_shared_vertices - start
[2019-11-18 13:53:31.380048] [0x000000010eeee5c0] [trace]   TriangleMeshSlicer::require_shared_vertices - stl_generate_shared_vertices
[2019-11-18 13:53:31.380065] [0x000000010eeee5c0] [trace]   TriangleMeshSlicer::require_shared_vertices - end
[2019-11-18 13:53:32.086635] [0x000000010eeee5c0] [info]    Checking for cached configuration updates...
[2019-11-18 13:53:32.095155] [0x000000010eeee5c0] [debug]   Vendor: Prusa Research, version installed: 1.0.3, version cached: 1.0.4
[2019-11-18 13:53:32.097163] [0x000000010eeee5c0] [info]    Downloaded index for `PrusaResearch` is the same as installed one, not offering an update.
[2019-11-18 13:53:32.097286] [0x000000010eeee5c0] [info]    No configuration updates available.
[2019-11-18 13:53:32.147922] [0x000070000e74c000] [info]    Downloading PrusaSlicer online version from: `https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version`
*   Trying 217.16.180.252...
* TCP_NODELAY set
* Connected to files.prusa3d.com (217.16.180.252) port 443 (#0)
* TLS 1.2 connection using TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384
* Server certificate: files.prusa3d.com
* Server certificate: Let's Encrypt Authority X3
* Server certificate: DST Root CA X3
> GET /wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version HTTP/1.1
Host: files.prusa3d.com
User-Agent: PrusaSlicer/2.1.0
Accept: */*

< HTTP/1.1 200 OK
< Server: nginx
< Date: Mon, 18 Nov 2019 18:53:32 GMT
< Content-Length: 6
< Connection: keep-alive
< X-Accel-Version: 0.01
< Last-Modified: Mon, 18 Nov 2019 18:01:05 GMT
< ETag: "6-597a2bb39439e"
< Accept-Ranges: bytes
< X-Powered-By: PleskLin
< 
* Connection #0 to host files.prusa3d.com left intact
[2019-11-18 13:53:32.707292] [0x000070000e74c000] [info]    Got PrusaSlicer online version: `2.1.0`. Sending to GUI thread...
[2019-11-18 13:53:32.709287] [0x000070000e74c000] [info]    Syncing configuration cache
[2019-11-18 13:53:32.709347] [0x000070000e74c000] [info]    Downloading index for vendor: Prusa Research
[2019-11-18 13:53:32.709416] [0x000070000e74c000] [info]    Get: `http://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaResearch//index.idx`
	-> `/Users/ninja.n/Library/Application Support/PrusaSlicer/cache/PrusaResearch.idx-update`
	via tmp path `/Users/ninja.n/Library/Application Support/PrusaSlicer/cache/PrusaResearch.idx-update.43586.download`
*   Trying 217.16.180.252...
* TCP_NODELAY set
* Connected to files.prusa3d.com (217.16.180.252) port 80 (#0)
> GET /wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaResearch//index.idx HTTP/1.1
Host: files.prusa3d.com
User-Agent: PrusaSlicer/2.1.0
Accept: */*

< HTTP/1.1 200 OK
< Server: nginx
< Date: Mon, 18 Nov 2019 18:53:32 GMT
< Content-Type: application/octet-stream
< Content-Length: 5690
[... 1889 lines cut ...]
[2019-11-18 13:54:16.948505] [0x000070000eb4f000] [trace]   Generating perimeters for layer 510
[2019-11-18 13:54:16.948515] [0x000070000eb4f000] [trace]   Generating perimeters for layer 510, region 0
[2019-11-18 13:54:16.950039] [0x000070000f758000] [trace]   Generating perimeters for layer 173 - Done
[2019-11-18 13:54:16.950059] [0x000070000f758000] [trace]   Generating perimeters for layer 174
[2019-11-18 13:54:16.950066] [0x000070000f758000] [trace]   Generating perimeters for layer 174, region 0
[... cut ...]
[2019-11-18 13:54:16.956378] [0x000070000f758000] [trace]   Generating perimeters for layer 176, region 0
[2019-11-18 13:54:16.958489] [0x000070000eb4f000] [trace]   Generating perimeters for layer 511 - Done
[2019-11-18 13:54:16.958512] [0x000070000eb4f000] [trace]   Generating perimeters for layer 512
[2019-11-18 13:54:16.958524] [0x000070000eb4f000] [trace]   Generating perimeters for layer 512, region 0
[2019-11-18 13:54:16.959025] [0x000070000f758000] [trace]   Generating perimeters for layer 176 - Done
[2019-11-18 13:54:16.959045] [0x000070000f758

lukasmatena, 2019-11-19:
@norre-daroy Thank you very much. The log indicates it crashed during infill generation, which may be a sign of more bugs in recently reworked gyroid infill calculation. You can try to verify  that by turning the infill off (Print Settings->Infill density - set to 0%) or setting different infill than gyroid.

Could you please enclose a 3MF (File->Save As) so we can try to reproduce the issue here?  (The 3MF contains the model and all config.) Thanks.

norre-daroy, 2019-11-20:
@lukasmatena  It didn't crash when I turned off the infill and changed the infill gyroid. I also enclosed the 3MF. However, I also noticed that it hangs when 'generating g-code' in the terminal

[norre-ninja-prusaSlicer.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3870601/norre-ninja-prusaSlicer.3mf.zip)

lukasmatena, 2019-11-21:
@norre-daroy Thanks. It does not crash on my Ubuntu 16.04, but @FidelCapo managed to reproduce it on OSX. I enclose an ASAN report that he provided for reference.

Probably @bubnikv will be interested.

Note: it is necessary to turn on gyroid infill after loading the 3MF.

<details>
<summary>ASAN report</summary>
==55892==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x603000c80d60 at pc 0x00010fc5b88e bp 0x7000052d65f0 sp 0x7000052d65e8
READ of size 8 at 0x603000c80d60 thread T17
    #0 0x10fc5b88d in Slic3r::mark_boundary_segments_touching_infill(std::__1::vector<std::__1::vector<Slic3r::Point, std::__1::allocator<Slic3r::Point> >, std::__1::allocator<std::__1::vector<Slic3r::Point, std::__1::allocator<Slic3r::Point> > > > const&, std::__1::vector<std::__1::vector<Slic3r::ContourPointData, std::__1::allocator<Slic3r::ContourPointData> >, std::__1::allocator<std::__1::vector<Slic3r::ContourPointData, std::__1::allocator<Slic3r::ContourPointData> > > >&, Slic3r::BoundingBox const&, std::__1::vector<Slic3r::Polyline, std::__1::allocator<Slic3r::Polyline> > const&, double, double)::Visitor::operator()(int, int) (PrusaSlicer:x86_64+0x10029888d)
    #1 0x10fc52c17 in void Slic3r::EdgeGrid::Grid::visit_cells_intersecting_line<Slic3r::mark_boundary_segments_touching_infill(std::__1::vector<std::__1::vector<Slic3r::Point, std::__1::allocator<Slic3r::Point> >, std::__1::allocator<std::__1::vector<Slic3r::Point, std::__1::allocator<Slic3r::Point> > > > const&, std::__1::vector<std::__1::vector<Slic3r::ContourPointData, std::__1::allocator<Slic3r::ContourPointData> >, std::__1::allocator<std::__1::vector<Slic3r::ContourPointData, std::__1::allocator<Slic3r::ContourPointData> > > >&, Slic3r::BoundingBox const&, std::__1::vector<Slic3r::Polyline, std::__1::allocator<Slic3r::Polyline> > const&, double, double)::Visitor>(Slic3r::Point, Slic3r::Point, Slic3r::mark_boundary_segments_touching_infill(std::__1::vector<std::__1::vector<Slic3r::Point, std::__1::allocator<Slic3r::Point> >, std::__1::allocator<std::__1::vector<Slic3r::Point, std::__1::allocator<Slic3r::Point> > > > const&, std::__1::vector<std::__1::vector<Slic3r::ContourPointData, std::__1::allocator<Slic3r::ContourPointData> >, std::__1::allocator<std::__1::vector<Slic3r::ContourPointData, std::__1::allocator<Slic3r::ContourPointData> > > >&, Slic3r::BoundingBox const&, std::__1::vector<Slic3r::Polyline, std::__1::allocator<Slic3r::Polyline> > const&, double, double)::Visitor&) const (PrusaSlicer:x86_64+0x10028fc17)
    #2 0x10fc52586 in Slic3r::mark_boundary_segments_touching_infill(std::__1::vector<std::__1::vector<Slic3r::Point, std::__1::allocator<Slic3r::Point> >, std::__1::allocator<std::__1::vector<Slic3r::Point, std::__1::allocator<Slic3r::Point> > > > const&, std::__1::vector<std::__1::vector<Slic3r::ContourPointData, std::__1::allocator<Slic3r::ContourPointData> >, std::__1::allocator<std::__1::vector<Slic3r::ContourPointData, std::__1::allocator<Slic3r::ContourPointData> > > >&, Slic3r::BoundingBox const&, std::__1::vector<Slic3r::Polyline, std::__1::allocator<Slic3r::Polyline> > const&, double, double) (PrusaSlicer:x86_64+0x10028f586)
    #3 0x10fc551e4 in Slic3r::Fill::connect_infill(std::__1::vector<Slic3r::Polyline, std::__1::allocator<Slic3r::Polyline> >&&, Slic3r::ExPolygon const&, std::__1::vector<Slic3r::Polyline, std::__1::allocator<Slic3r::Polyline> >&, Slic3r::FillParams const&) (PrusaSlicer:x86_64+0x1002921e4)
    #4 0x10fc6ba84 in Slic3r::FillGyroid::_fill_surface_single(Slic3r::FillParams const&, unsigned int, std::__1::pair<float, Slic3r::Point> const&, Slic3r::ExPolygon&, std::__1::vector<Slic3r::Polyline, std::__1::allocator<Slic3r::Polyline> >&) (PrusaSlicer:x86_64+0x1002a8a84)
    #5 0x10fc51125 in Slic3r::Fill::fill_surface(Slic3r::Surface const*, Slic3r::FillParams const&) (PrusaSlicer:x86_64+0x10028e125)
    #6 0x10fc45093 in Slic3r::Layer::make_fills() (PrusaSlicer:x86_64+0x100282093)
    #7 0x1103821ec in tbb::interface9::internal::start_for<tbb::blocked_range<unsigned long>, Slic3r::PrintObject::infill()::$_3, tbb::auto_partitioner const>::execute() (PrusaSlicer:x86_64+0x1009bf1ec)
    #8 0x111a036a8 in tbb::internal::custom_scheduler<tbb::internal::IntelSchedulerTraits>::local_wait_for_all(tbb::task&, tbb::task*) (PrusaSlicer:x86_64+0x1020406a8)
    #9 0x1119f7059 in tbb::internal::arena::process(tbb::internal::generic_scheduler&) (PrusaSlicer:x86_64+0x102034059)
    #10 0x1119fe424 in tbb::internal::market::process(rml::job&) (PrusaSlicer:x86_64+0x10203b424)
    #11 0x1119ff437 in tbb::internal::rml::private_worker::run() (PrusaSlicer:x86_64+0x10203c437)
    #12 0x1119ff3d8 in tbb::internal::rml::private_worker::thread_routine(void*) (PrusaSlicer:x86_64+0x10203c3d8)
    #13 0x7fff73b2d2ea in _pthread_body (libsystem_pthread.dylib:x86_64+0x32ea)
    #14 0x7fff73b30248 in _pthread_start (libsystem_pthread.dylib:x86_64+0x6248)
    #15 0x7fff73b2c40c in thread_start (libsystem_pthread.dylib:x86_64+0x240c)
</details>

closed 2026-09-29 by prusaslicer-bot[bot]