prusa3d/PrusaSlicer #3183: TBB library cannot be found. Consider set TBBROOT environment variable.

norre-daroy, 2019-11-13:
### Version
version_2.1.0-485-g050a9adf3

### Operating system type + version
macOS Mojave 10.14.6

### Behavior
* _Describe the problem_
I am new to PrusaSlic3r and bit rusty with C++. I git clone `https://github.com/prusa3d/PrusaSlicer.git`. I then went into `deps` directory. After I `mkdir build` and `cd build`, I get this error 
> TBB library cannot be found. Consider set TBBROOT environment variable.
  (missing: TBB_INCLUDE_DIRS TBB_LIBRARIES)

Any help is appreciated. Thank you so much :)

[image]

_Is this a new feature request?_
No

tamasmeszaros, 2019-11-13:
Its hard to tell from this error message only. Check the CMake prefix path and if TBB is built properly there (should be a static library). Also if you use the deps directory to build the dependencies, and then configure PrusaSlicer project, make sure you have `-DSLIC3R_STATIC=ON` when calling cmake.

norre-daroy, 2019-11-13:
Hello . Thank you so much for responding. Checking the CMake prefix path, did you mean this command `cmake .. -DCMAKE_PREFIX_PATH="$PWD/../deps/build/destdir/usr/local"`? Because after git clone, all I did was 

`mkdir build`
`cd build`
`cmake ..`
and then I got this error. How do I build TBB?

norre-daroy, 2019-11-13:
Hello. Thank you so much for replying. When you said check the CMake prefix path is it this command `cmake .. -DCMAKE_PREFIX_PATH="$PWD/../deps/build/destdir/usr/local"`. Because all I did was build the dependencies as per instructed in the `deps` directory with `cmake ..` then `make`. And I got some `make` errors.
[image]

But I ignored these errors and proceeded to build PrusaSlicer with `mkdir build`, `cd build`, `cmake .. -DCMAKE_PREFIX_PATH="$PWD/../deps/build/destdir/usr/local"`. What step am I missing and where should I add`-DSLIC3R_STATIC=ON` when calling cmake?

In this screenshot, it says my SLIC3R_STATIC is already ON. But CMAKE_PREFIX_PATH element does not exist.
[image]

 Thank yo so much for your help

lukasmatena, 2019-11-13:
@norre-daroy Your deps build failed on building boost, so it never got to build TBB. That's why it wasn't found. Boost was, because you have it installed on your system.

As to why it failed, I don't know. There is something wrong in the path around the `WebDev` directory. Is there some chance it has a trailing space at the end? If so, I'd repeat what you were doing in e.g. `/Users/ninja.n/PrusaSlicer` to see what happens.

The `-DSLIC3R_STATIC=ON` should be passed as another argument to cmake when building PrusaSlicer (not the dependencies).

norre-daroy, 2019-11-13:
@lukasmatena you were right. I had some trailing space at the end. I successfully built it.

closed 2019-11-13 by norre-daroy

italocjs, 2019-12-22:
Hi, how did you fix this? i'm stuck in a similar error, i never used visual studio before and i'm totally lost

vitasimek, 2019-12-30:
I have same problem on Debian Bullseye.

senseisimple, 2020-05-14:
> I have same problem on Debian Bullseye. TBB installed by apt.

If it's installed by apt, you'll probably need to add ` -DTBB_LIBRARIES=/usr/include/tbb` to your cmake command.

i.e. `cmake .. -DSLIC3R_STATIC=1 -DCMAKE_PREFIX_PATH=/opt/PrusaSlicer -DTBB_LIBRARIES=/usr/include/tbb`

boromyr, 2020-09-27:
I am also stuck with the tbb libraries, I keep getting this error from CMake all the time, despite having entered the correct path. Latest stable release downloaded from https://github.com/oneapi-src/oneTBB, but there is no instruction on how to install it.

italocjs, 2020-09-27:
its been quite some time i fixed this, but i think i did a fresh install on C:/ instead of default directory.

boromyr, 2020-09-29:
Should I then install boost and tbb in C:\?

Drachenkaetzchen, 2021-02-22:
If you are on Windows, use absolute paths. I stumbled on this when I tried to be smart and used a relative path like `..\..\PrusaSlicer-deps\usr\local`, which resulted in this error. A full path like `C:\src\PrusaSlicer-deps\usr\local` did work.

spuder, 2021-12-05:
After fighting this for hours on OSX 11.6, I finally worked around it by resetting the `deps` directory in git. 

```
rm -rf deps/build
rm -rf deps
git checkout deps/*
mkdir -p deps/build
cd deps/build
cmake ..
```