prusa3d/PrusaSlicer #3184: straw stl ends up closed at both ends

iplayfast, 2019-11-13:
### Version
2.1.0+Linux64

### Operating system type + version
Linux Mint 19.2
### 3D printer brand / version + firmware version (if known)
Prusa i3 Mk3
### Behavior
* simple straw stl ends up being closed at both ends and partially filled.
* _Steps needed to reproduce the problem_
Load and slice attached stl
zip contains straw.stl and straw.gcode
[straw.zip](https://github.com/prusa3d/PrusaSlicer/files/3841970/straw.zip)

Jebtrix, 2019-11-13:
Huh? Its a cylinder not a hollow tube..

[image]

lukasmatena, 2019-11-15:
This is indeed a cylinder, not a straw. As far as I can tell, PrusaSlicer does what it should.

Closing the issue. If you object, I can reopen it.

closed 2019-11-15 by lukasmatena