prusa3d/PrusaSlicer #3185: PrusaSlicer hangs on the "generating g-code" step when slicing large files.

inventaholic, 2019-11-13:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

Windows 10 pro 64 bit

_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)

artillery x1

_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior

prusaslicer hangs at the "generating gcode" step

[grip_roller_point05_05.zip](https://github.com/prusa3d/PrusaSlicer/files/3842119/grip_roller_point05_05.zip)

 when slicing large files

* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Jebtrix, 2019-11-13:
Avoid crossing perimeters option adds significant processing time but does finish. Latest git master was roughly 27% faster on old i7 920 CPU.
* 2.1.0 - 7m 34s
* git master - 5m 30s

bubnikv, 2020-12-11:
"Avoid crossing perimeters" has been reworked with 5f456ebbe78f3b994ea7ecc7de23c119e4705b61 (see the branch lh_avoid_crossing_perimeters).
The new algorithm will be available in PrusaSlicer-2.3.0-beta3. Please check the PrusaSlicer-2.3.0-beta3 change log for details.

[image]

closed 2020-12-11 by bubnikv

well-engineered, 2020-12-14:
"bug" still present for me on PrusaSlicer-2.3.0-beta3.

hejllukas, 2020-12-14:
Does it hang on the same 3MF file attached in this issue or a different one? If you try to disable Avoid crossing perimeters, it is still hanging or not?

inventaholic, 2020-12-14:
This never went away for me either. ANy time I have a complex layout with
10 or more parts on the bed and "don't cross perimeters" is checked, it
never completes.

well-engineered, 2020-12-14:
For me, on mac, 2.3.0beta3, it is on every "a bit more complex" printjob aka multicolor or multiple parts. I can reproduce this on demand easily.

hejllukas, 2020-12-14:
Ok, so could you please provide a 3MF file? We could try to reproduce it.

well-engineered, 2020-12-14:
here is one I have on hand atm: https://1drv.ms/u/s!AjQRNRGFeANNv2TR6z--eHZraP8Q?e=2G9TiR

well-engineered, 2020-12-14:
Oh, probably worth to mention:
I did run this on a maxed out 2020 MacBook Pro, Intel I9, 32GB RAM etc and let it run over night, because I thought, maybe it will need "some time". It ran for the night and the day I was at work, so probably 16h and wasn't finished when I got home in the afternoon.

hejllukas, 2020-12-15:
Thank you for providing your 3MF file and information about your computer. We tried to reproduce your issue on Windows, Linux, and also Mac OS X with PrusaSlicer 2.3.0-beta3. But on all platforms, it takes only a few minutes to slice with enabled avoid crossing perimeters. Did you really try it in [PrusaSlicer 2.3.0-beta3](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-beta3)? Because all previous versions, including PrusaSlicer 2.3.0-beta2, have an issue with very long slicing or hanging, but PrusaSlicer 2.3.0-beta3 should fix this issue.

well-engineered, 2020-12-15:
**I re-downloaded beta3** just to make sure about it and used it to slice said 3MF-File and now **it works**. 
Thank you for taking a look at it!
I don't know what happened before or why it didn't work out. Maybe I didn't use the beta3, but thought I did?
Sorry for the inconvenience I might have made.