prusa3d/PrusaSlicer #3186: 'NLopt Library' not found while compiling Slic3r Prusa on Windows

coyote-seized, 2019-11-13:
Hi,

I am running the following code on VS command prompt as directed on the website, https://github.com/prusa3d/PrusaSlicer/wiki/%5BLegacy%5D-How-to-compile-Slic3r-Prusa-Edition-on-MS-Windows

`cmake .. -G "NMake Makefiles" -DCMAKE_BUILD_TYPE=Release`

Below is the error message that I am receiving,

```
Unable to find requested NLopt installation: Cannot find NLopt library
  'nlopt_cxx' in 'C:\local\Nlopt/lib'.
Call Stack (most recent call first):
  CMakeLists.txt:399 (find_package)

-- Configuring incomplete, errors occurred!
See also "C:/local/Slic3r/build/CMakeFiles/CMakeOutput.log".
See also "C:/local/Slic3r/build/CMakeFiles/CMakeError.log".

```

I downloaded NLopt for this website, 
https://nlopt.readthedocs.io/en/latest/NLopt_on_Windows/

Below is the path that I defined for the lib directory, not sure of what the problem is, would really appreciate if someone can help.
NLopt_INCLUDE_DIR - C:\local\Nlopt\include
NLopt_LIBRARY_DIR - C:\local\Nlopt\lib

coyote-seized, 2019-11-13:
Renaming the lib file to 'nlop_cxx.lib' solved the issue!

closed 2019-11-13 by coyote-seized

Jebtrix, 2019-11-13:
The cmake project will pull nlopt for you. Your making more work for yourself.