prusa3d/PrusaSlicer #3188: How to extend PrusaSlicer with scripts or Plugins?

X3msnake, 2019-11-14:
How can one write a plugin or script for PrusaSlicer i dont find documentation about it anywhere.

I want to make a plugin to export the SLA slices to CWS, cbddlp, photon and try to integrate the slicer with some of the photonsters tools like Validator

i know we can change the extension on export, but at the moment MSLA slicing does not add any gcode 

One also need the Gcode to be formatted in specific ways

tamasmeszaros, 2019-11-15:
Hello! 
There is no such thing in PrusaSlicer as plug-in bundles. If you need to extend it beyond the boundaries of custom profiles, grab the source code and make it happen. We also appreciate pull requests.

X3msnake, 2019-12-10:
@tamasmeszaros where can i learn about the slicing logic and files PrusaSlicer uses to create the slices and write the images plust the config file to the .sl1 zip file?

tamasmeszaros, 2020-03-16:
@X3msnake, Sorry for the late reply. There is no documentation other than the source code and its comments. You are right in that the SLA story resides in src/libslic3r/SLA folder. These form the backend for SLA print preparation. However, the actual slicing is done in src/libslic3r/SLAPrint.cpp and src/libslic3r/SLAPrintSteps.cpp. There is a class Zipper (Zipper.hpp) which is handed over to the SLAPrint object when the output archive is constructed. You can use that zipper object to write additional files into the output archive. See line 160 in src/slic3r/GUI/BackgroundSlicingProcess.cpp.

Good luck!

tamasmeszaros, 2020-03-16:
This topic is also touched in #3713

closed 2020-03-16 by tamasmeszaros

X3msnake, 2020-03-16:
Thanks for the update and the info :)

16/03/2020 à(s) 08:51:

simonjohansson, 2020-05-29:
@X3msnake did you get anywhere with this?

X3msnake, 2020-05-29:
we ended up creating a command line conversion tool.

lookup uv3dp