prusa3d/PrusaSlicer #3189: min_fan_speed not accessible in Printer Settings > Custom G-code

zehkaiser, 2019-11-14:
### Version
PrusaSlicer 2.1.0 win64 201909160915

### Operating system type + version
Windows 10 Pro 1803

### Behavior
When using the placeholder min_fan_speed PrusaSlicer will throw an error on gcode creation.

After layer change G-code:
`{if layer_num == 1}{if min_fan_speed < 255}M106 S40{endif}`

Produced file:
`!!!!! Failed to process the custom G-code template before_layer_gcode
Parsing error at line 5: Referencing a vector variable when scalar is expected
{if layer_num == 1}{if min_fan_speed < 255}M106 S40{endif}
                       ^
!!!!! End of an error report for the custom G-code template before_layer_gcode`

Jebtrix, 2019-11-14:
I see two **_if_** statements and only one **_endif_** closure unless there is part of the custom code missing in the post.

The **_min_fan_speed_** is an array type (vector) placeholder because of multi-material where you have multiple filaments in play that can have different fan speeds for each.

`{if layer_num == 1 and min_fan_speed[current_extruder] < 255}M106 S40{endif}`

zehkaiser, 2019-11-14:
Yes, the missing `{endif}` was just a copy-paste error. The index is what I was missing. I didn't realize that min_fan_speed is an array (I tried min_fan_speed_0, but not without the underscore). Thanks for the help.

closed 2019-11-14 by zehkaiser

Jebtrix, 2019-11-14:
No problem. You can reference the wiki for more details:

https://github.com/prusa3d/PrusaSlicer/wiki/Slic3r-Prusa-Edition-Macro-Language