prusa3d/PrusaSlicer #3192: [2.1.0+-201909160907 - MacOS] Unsupported command in Prusa FW: G21 , still included in MMU system profiles

cryocide, 2019-11-15:
### Version
2.1.0+-201909160907

### Operating system type + version
MacOS 10.15.1

### 3D printer brand / version + firmware version (if known)
Original Prusa I3 MK3S MMU2S

### Behavior
* G21 command is included in Custom G-code section of MMU/MMU2 system profiles
* Add new printer via wizard, check any MMU profile, including MMU2. View printer settings tab, custom G-code section. Scroll past G80 command to find G21.
* G21 should not be present
* G21 is present, and current (3.8.1) Prusa firmware gives an "unknown gcode" error when this command is read.
[image]

lukasmatena, 2019-11-19:
@rtyr is looking into it.

rtyr, 2019-12-02:
We will remove it from our profiles. Thank you,

lukasmatena, 2019-12-17:
The G21 gcode was removed from our profiles starting with version 1.0.6.
Thanks for pointing it out. Closing.

closed 2019-12-17 by lukasmatena