prusa3d/PrusaSlicer #3193: Included boost fails to build on CentOS7

MakerMatrix, 2019-11-15:
### Version
version_2.1.0-503-gb2a6a07

### Operating system type + version
CentOS7

### Behavior
Included boost-nowide does not build due to dependency failure (missing include files)

47 scylla:/home/jsmith/prusa/PrusaSlicer% cd src/
348 scylla:/home/jsmith/prusa/PrusaSlicer/src% cd boost
349 scylla:/home/jsmith/prusa/PrusaSlicer/src/boost% cmake3 .
-- The C compiler identification is GNU 4.8.5
-- The CXX compiler identification is GNU 4.8.5
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jsmith/prusa/PrusaSlicer/src/boost
350 scylla:/home/jsmith/prusa/PrusaSlicer/src/boost% make -j4
Scanning dependencies of target nowide
[ 50%] Building CXX object CMakeFiles/nowide.dir/nowide/iostream.cpp.o
/home/jsmith/prusa/PrusaSlicer/src/boost/nowide/iostream.cpp:9:37: fatal error: boost/nowide/iostream.hpp: No such file or directory
 #include <boost/nowide/iostream.hpp>
                                     ^
compilation terminated.
make[2]: *** [CMakeFiles/nowide.dir/nowide/iostream.cpp.o] Error 1
make[1]: *** [CMakeFiles/nowide.dir/all] Error 2
make: *** [all] Error 2
351 scylla:/home/jsmith/prusa/PrusaSlicer/src/boost%

MakerMatrix, 2019-11-16:
Fix is to remove the leading path for the #include lines in iostream.cpp and iostream.hpp

vojtechkral, 2019-11-16:
Why are you making just the boost/nowide subdir? That's not going to work, because correct include paths are set in the parent CMakeLists.txt. It's meant to be built as part of the whole slicer.

MakerMatrix, 2019-11-16:
Because the whole slicer wouldn't build and that was the first error that make produced.  One thing at a time ;-)

closed 2019-11-16 by MakerMatrix

vojtechkral, 2019-11-16:
> I'm fairly sure that whoever put that in the tree had the boost libs/includes installed at the normal system locations and so they didn't notice this was broken.

I'm fairly sure that's not the case. The `boost` subdir doesn't contain boost, it only contains the nowide headers, which aren't part of boost (as far as I know, anyway) and in any case I don't have the nowide headers anywhere on my system (and yet slicer builds).

I feel like this is an X/Y problem. You should either post or inspect more thoroughly your original error, it was probably unrelated to the `boost/nowide` directory in the tree.

MakerMatrix, 2019-11-16:
You're right.  The problem is with the build instructions.  Making a build/ subdir and doing "cmake .." breaks the relative paths in a lot of things.