prusa3d/PrusaSlicer #3194: Print bed apperence has changed back

Tulle42, 2019-11-16:
### Version
_Version of PrusaSlicer used goes here_
2.1.0 win64
_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
windows 10 Pro ver. 1909 OS 18363.476
### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Prusa mk3s firmware 3.8.1
### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
After updating windows the apperence of the print area/bed has changed back to the basic square and not the prusa printer view.
[image]

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Jebtrix, 2019-11-16:
Did you at least try reinstalling GFX drivers yet?

Tulle42, 2019-11-16:
Yes, I have tried reinstalling and uppdating GFX drivers und everything that commes with prusaslicer.

bubnikv, 2020-03-15:
is this issue solved for you already? I suppose it is abroken configuration issue.

rtyr, 2020-11-16:
No response. Closing.

closed 2020-11-16 by rtyr