prusa3d/PrusaSlicer #3198: Windows installer doesn't wait for user seleciton for reboot; reboots immediately

robmikh, 2019-11-17:
### Version
2.1.0+win64

### Operating system type + version
Windows 10 (build 18362), 64-bit

### 3D printer brand / version + firmware version (if known)
Irrelevant, this is during setup without a printer hooked up to the computer.

### Behavior
* Download the Windows installer from the prusa3d website
* Run the installer

Current behavior:
When the installer gets to the page where it asks the user whether or not to restart, the computer immediately starts to reboot. It happened so quickly I didn't see that the installer was on that page already (I was able to cancel the restart and see the installer was asking me about rebooting).

Expected behavior:
Wait for the user to make a selection before rebooting, and only if the user chooses to do so.

This was very disruptive, as I was doing other things at the same time.

#### Project File (.3MF) where problem occurs
No print file used, this was during installation.

jokohoko, 2019-12-12:
I had the same effect a moment ago when updating to the 2.1.1+win64 version

Ark-kun, 2020-02-19:
Same problem on all machines.

Where is the installer config/code? I could not find it in this repo.

lukasmatena, 2020-03-04:
New installer that will be used for 2.2.0 final should not require a reboot. This should solve this issue.

Thanks for reporting. Closing.

closed 2020-03-04 by lukasmatena