prusa3d/PrusaSlicer #3199: [Feature Request] Per Minute Cost

MINT3D, 2019-11-18:
An alteration to the pricing algorithm to add a rate per minute. Pricing algorithm would multiply by minutes to print and add to the material cost.

Pricing algorithm becomes:
Cost = (filament used (g) x filament cost (/kg)) + (rate per minute x minutes to print  )

This  would allow people to price based on time, material, combination of both. If left blank, price is material cost only.

While you can do this to some extent by inflating the /kg cost of filament, time can vary hugely for models with identical weight based on complexity.

wes1007, 2020-03-28:
I would also like to see this implemented.

It may also be more beneficial to use cost per hour rather than per minute as users could get a per hour cost of running their machine which might be more accurate when combined with a "setup cost". Setup cost could include electricity usage when starting a print from cold 

Id like to add that i think this feature should be added to the printer profile as it could then be used to factor in electricity usage per minute/hour for the different printers that one may have. 

This would allow users who print for profit to add a markup to their filament cost and then add a markup to the print time costs as well as setup costs. It would also allow hobbyist to get a better idea of the cost of printing their part.

Another option would be allowing the user to input a cost per KWH (Under Printer Profile as it would depend on the printer), startup time (the time for the machine to get to operating temperature) (Under Filament Profile) and a profit/markup per hour(under Printer Profile). This would allow the user to adjust the cost of electricity without having to rerun all their calculations if the rate happens to change at a later date.

fiveangle, 2021-03-08:
Was going to submit FR for estimated energy costs to be included in the slicing statistics but searching for existing FR I found this one which seems to cover at least some of this.

My desire is to simply have a more accurate estimated cost.  At $0.52/kWh for peak power usage time here from our corrupt PG&E power provider, it is not an insignificant cost when you add it all up.

Thanks for considering this addition !

DeadCatAlive, 2021-03-24:
Prusa already has [this great price calculator](https://blog.prusaprinters.org/3d-printing-price-calculator_38905/), it'd be awesome if it could be available as a popup window or similar on the model view, with the fields for the labor cost to fill out. Electricity and upkeep cost could be added to the printer settings, and print time, filament type, filament price and weight of the print are already available in PrusaSlicer.

closed 2026-09-29 by prusaslicer-bot[bot]