prusa3d/PrusaSlicer #3201: App crashes / hangs when non-numeric characters are typed in numeric fields.

gryn, 2019-11-18:
### Version
2.1.0

### Operating system type + version
OS X

### 3D printer brand / version + firmware version (if known)
Prusa

### Behavior
Numeric entry boxes, such as scale and position, cause application failure if a non-numeric field is typed. Application hangs and must be force quit. To reproduce try typing # in the size field.

[image]

JohnOCFII, 2019-11-18:
Interesting.  When I try that, I get a standard error dialog box.  Also macOS Catalina.
[image]

gryn, 2019-11-18:
I've had this behavior for a while it's not a 2.1 bug.

Here is my OS X version perhaps it's related.

[image]

FidelCapo, 2020-01-09:
Thank you for report. This issue should be fixed in PrusaSlicer 2.2.0-alpha2.
Closing.

closed 2020-01-09 by FidelCapo