prusa3d/PrusaSlicer #3202: Slicer producing support interface layer overhangs

peterson-caleb, 2019-11-18:
### Version
2.1.0+win64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10 x64
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
Stock Prusa MK3S - 3.8.0

### Behavior
Slicer is leaving an overhang on the interface layers of some support material. This causes the support to often fail causing the whole print to fail. It only happens on one side of a single part that I have designed. The part is symmetrical side to side.
[image]

[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
[body16410663.zip](https://github.com/prusa3d/PrusaSlicer/files/3859680/body16410663.zip)

Aquaerics, 2020-01-09:
I saw that too when i slice similar designs. But i have an overhang on two sides if it is a symetrical object.

ParkyCode, 2020-03-26:
I think this shows the mysterious added overhangs.  I wanted to stack my prints.  I created 2 and left a Z gap.  I then created a support enforcer of the part, and fitted it so it covered the Z gap.

[image]

I wanted just the support material in the gap (support material interface, dark green) and not around the outside of the part (support material, light green).  I'm not sure why its creating the light green section.

[image]

Is there a way to turn off the "support Material" (light green), but keep the "support material interface" (dark green)?

Crawlerin-SA, 2020-05-12:
Hello, I encountered the same issue when trying to slice dual fan duct model (it's Bullseye fanduct for ender 3). It generates support interfaces mid-air at exhausts, both when it's oriented by X or Y axis.

It happens with 0.15 mm layer height for me, both quality and speed profiles, regardless of selected material. if I select 0.2 it slices without those support interfaces mid-air.

Version: 2.2.0+win64
Build: PrusaSlicer-2.2.0+win64-202003211145
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 18363), 64-bit edition
Total RAM size [MB]: 17,109MB

[image]
[image]

Aquaerics, 2020-11-12:
Anny news about this bug? would this problem be fixed? I had similar problems wenn i try to support overhangs.

bubnikv, 2021-02-22:
I actually was not able to reproduce the issue with PrusaSlicer 2.3.0, however the following issues demonstrate the same:

Support Interface printed on air #2903
Supports generated loose in space #4874
[2.3.0-alpha2] Support Interface layers in midair #5016

and these issues were fixed with 3985d7bb81ddd542612ec2e8704622be4e90f074

closed 2021-02-22 by bubnikv