prusa3d/PrusaSlicer #3204: Feature Request: Stop vase mode and revert to standard mode at set height

pieroway, 2019-11-18:
New feature request.  Be able to combine vase mode with non-vase mode.
I'd like to specify vase mode up to a certain height then revert to non-vase mode for rest of print.  
I personally can use this for creating bottles with a stronger spout.  Clean vase mode perimeter up to where the bottle neck begins then it start to allow for more infill and multi-perimeter walls.

cooperised, 2021-07-28:
+1.  I'd like the opposite to the OP, but the same feature: normal mode for the base of a box with ribs, and then vase mode for the rest.  Vase mode on a modifier would be great.

greenveg, 2022-02-22:
+1.
Just make vase mode accessible from modifiers, be it volumes/bodies or height.

I need a sturdy bottom...

James-G4CLF, 2022-06-06:
I quite often need shaped holes in the bottom of vase structures, and vase printed screw-top bottles would be stonger and lighter, so add my support to this suggestion. #

spineytoad, 2022-11-10:
+1 for me also

dodasch, 2023-01-12:
need this feature, so i don't have to combine two gcodes! 

+2

hrvojebazina, 2023-01-19:
Vase mode completely messes up the layer with the letters as you can see. So this feature would definitely be necessary to add
[image]

gudvinr, 2023-02-13:
@thiebes there are reaction buttons for this purpose. With "+1" comments you (and everyone else who does this) only ping everyone subscribed to this issue. People subscribe to track progress and not for this.

thiebes, 2023-02-13:
Thanks for that feedback. Won’t happen again!

BloodyRain2k, 2023-04-25:
+1 because I just had a case where I wanted to print 2 parts that belong together, a capsule and it's cap, and was a little disappointed that Spiral Mode is not available as a per model setting

UmbraAtrox, 2023-07-04:
+1 I tried to split my part in prusaslicer to workaround but i also can't place it above Z0.00 ontop of the other part so idk why that coordinate even exists as a editable field.

spineytoad, 2023-11-13:
+1 I really needed this the other day!

Henk72, 2023-11-15:
+1 This would be awesome!

nils-a, 2023-11-15:
People, could you please stop adding "+1" comments?!

**There are reaction buttons for this purpose.** 

With "+1" comments, you only ping everyone subscribed to this issue. People subscribe to track progress and not to get notified of other people feeling the issue is more important than other issues.

kokoianer, 2024-02-12:
Hey Ho, good evening
It is possible to print the bottom part in non vase mode and the upper part in vase mode. (some asked for the opposite, but others may can use this as well).
Use the following setting:
Check Spiral Vase mode, allow all changes.
Now for the non spiral vase mode bottom, just use the following setting:
"Horizontale Konturhülle"->"Massive Schichten" -> "Boden" and enter the number of layers you want to have normal printed.
[image]

FoldZero, 2024-03-01:
I want to add an idea, i see the hot end jumping from one end of a print to the other a lot. 
This seems like a wasted trip as it goes back and forth.
Why can we not implement Vase Mode per layer?
One contiguous path for that layer, as its all the same height. This would result in no wasted Travel Moves.?

yahbluez, 2024-03-01:
> One contiguous path for that layer, as its all the same height. This would result in no wasted Travel Moves.?

That's not the way vase mode works. In vase mode the Z is increased in microsteps in a continously manner. At the end of one vase mode turn the nozzle is one layer height higher than at the beginning.

cooperised, 2024-03-06:
> It is possible to print the bottom part in non vase mode and the upper part in vase mode. (some asked for the opposite, but others may can use this as well). Use the following setting: Check Spiral [...]

That's not a solution to the problem at hand.  All that will do is print a number of solid layers at the bottom - no infill, no interior detail, nothing.

kristianmk, 2024-03-06:
For some limited use cases, having solid bottom layers will be sufficient, similar to this example: [https://github.com/kristianmk/party-hat-generator](https://github.com/kristianmk/party-hat-generator).

Allowing users to select the height for the transition to and from spiral mode would be a significant improvement. However, is an abrupt transition between normal and spiral mode always feasible? It is probably impossible for layers with support material active.

Figucp0, 2024-10-02:
I would love to see this as well, or at least understand the limitations that will not allow this.

azulez, 2024-12-07:
A use case I am running into for this is something that has multimaterial bottom layers, since vase mode is mutually exclusive to multi-material slicing, it seems.  Specifically the entire bottom thickness is flat and then, in an ideal world, the remaining hollow exterior should spiral upwards from there.

Vase spiral mode being able to be switched on and off during slicing at specific layers would be very helpful in this case as well.

Goury, 2025-02-01:
More than five years and not progress.
Shame.

James-G4CLF, 2025-02-01:
Maybe no progress on more flexible vase mode, which is indeed a pity as I would like to use it, but all sorts of other good stuff during that time so let's not be too critical.

If enough people were to march from their country's capital to Prague, each carrying a banner saying "Joe - sort out the more complex vase mode" in their own language:-
(Joe - rendezze a bonyolultabb váza módot.
ג'ו - סדר את מצב האגרטל המורכב יותר
जो - अधिक जटिल फूलदान मोड को सुलझाएं
etc.)

GTG3000, 2025-06-27:
Who else found this thread in 2025, hoping for some kinda solution to "just do the vase thing but from layer 51 instead"?

yahbluez, 2025-07-14:
@GTG3000 that already works, you can set the number of base layers in prusaslicer to 51 and start vase mode after that layer. Until the given base layer number all slicing options work as usual.

What I miss is this, "stop vase mode at layer xyz" and do again layer based slicing.

Goury, 2025-07-14:
@yahbluez it is not.
Prusa is a greedy corpo, he's just not interested in doing anything you may want.
And his employees couldn't be any less motivated.

Booyaka101, 2026-09-05:
I don't have a printer, which is a daft position to be posting this from, but here it
is: https://github.com/Booyaka101/vaseweld

You slice the plate twice, once with Spiral Vase off and once on, and it welds the two
files at a height you pick. Below the cut you keep actual perimeters and infill, so it
isn't the "just add more bottom solid layers" workaround. Two cuts and you get a solid
top as well.

It goes through Klipper's motion planner in batch mode without complaint, which is about
as far as I can take it on my own. Would anyone be up for running one and telling me what
the transition actually looks like?

greenveg, 2026-09-05:
This method does work. I’ve done successful prints with it. I just dont like the manual fuckery…

Booyaka101, 2026-09-15:
If you're splicing them by hand, you don't have to any more. 1.4.0 does the whole thing:

    pip install -U vaseweld
    vaseweld auto project.3mf --at 6.0 -o hybrid.gcode

It runs PrusaSlicer twice itself, once normal and once in vase mode, then welds the two.

I only caught this by diffing the two G-code files: ticking Spiral Vase in the GUI also flips six other settings, and it does that from a dialog that never runs headless. So a vase pass sliced from the command line with just --spiral-vase isn't the same toolpath the GUI gives you. 2.9.x only for now, 3.0 rewrote the argument parsing.

As far as I know nobody has actually printed one of the welded files yet. If you run one I'd really like to see the layer where the spiral starts.

wikolii, 2026-09-29:
I will try vaseweld, as i currently weld files manually on a weekly basis.
But this should still be a feature in the slicer