prusa3d/PrusaSlicer #3205: I'm so annoyed that PrusaSlicer 2.1.0 +win64 is crashing all the time. :(

e2mp, 2019-11-19:
I'm so annoyed that PrusaSlicer 2.1.0 +win64 is crashing all the time. :(
I downloaded and installed it last week. Is there a solution, anyone?
Thanks.
/Martin

### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Jebtrix, 2019-11-19:
You will need to provide more details for devs. Things to try:

* Did you upgrade from v2.0.0?
  * Rename folder `C:\users\<username>\appdata\roaming\prusaslicer` in case of problem with upgraded older profiles. This will force slicer to recreate default empty profile folder. 

* Open Command Prompt/Console and run: 
`"%programfiles%\Prusa3D\PrusaSlicer\prusa-slicer-console.exe" --loglevel=5 > "%UserProfile%\Desktop\prusa-debug-opengl-lvl5.log"`
  *  You can then attach log file for devs to review
* Open Command Prompt/Console and run slicer without GPU acceleration: 
`"%programfiles%\Prusa3D\PrusaSlicer\prusa-slicer-console.exe" --sw-renderer --loglevel=5 > "%UserProfile%\Desktop\prusa-debug-mesa-lvl5.log"`
  *  You can then attach log file for devs to review
* Try latest git version to see if problem has already been resolved:
  * [2.1.0-git-snapshot-win64](https://github.com/Jebtrix/PrusaSlicer/releases/tag/2.1.0-git-snapshot)

bubnikv, 2020-03-15:
no response, closing.

closed 2020-03-15 by bubnikv