prusa3d/PrusaSlicer #3207: NEW FEATURE REQUEST - Update Geometry / Object from originally imported file

SharifZaman, 2019-11-19:
It would be nice to have a simple update geometry button in PrusaSlicer to use as needed to re-import the geometry file to get the changes made and saved in the original CAD file.  It can be based on the original file name or as option, prompt for a new file name and location. 

This is typical in analysis software and CAM software that imports a CAD file for the geometry. 
 There is a way to update the imported geometry from the original or a new CAD file.  The CAM software will then compare and replace the object with the newly updated geometry.  Things like, position,  scale, and other object settings can be retained.  Some programs like MasterCam are very smart to to show the changes or it can be very basic and be developed with each revision of the software.
 
Sharif

Jebtrix, 2019-11-19:
There is already Reload from Disk option when your right click on a part. The upcoming v2.2 has added the feature to also work on saved projects.

SharifZaman, 2019-11-19:
Sweet!  Thanks for the very quick reply too!  I’m just amazed how fast you replied and the feature is there too.

I just tried it and seemed to work.  It did even retain the scale.  It did give me the warning again too large too fit before it resized.

Sharif

Jebtrix, 2019-11-19:
You will definitely enjoy the added features and refinement in that area in next release.

closed 2019-11-19 by vojtechkral