prusa3d/PrusaSlicer #3210: after slicing some of my screw holes are removed on the design

rugalash, 2019-11-20:
(see imgur pictures, it's pretty self explanatory) 
### Version
All of them and slic3r too
I use:
PrusaSlicer-2.1.0+linux64-201909160925.AppImage
However I sent my design to 3 friends and they also have the issue with slic3r and their prusaslicer, they use windows.

### Operating system type + version
Not sure that matters due to multiple testing above

### 3D printer brand / version + firmware version (if known)
I built my printer which isn't relevant to slicing software

### Behavior
We take the file my file, it does it in .stl, .amf, .obj, all the same. Never happened before I've been using freeCAD and slic3r / prusaslicer for several years now.
How to reproduce you simply import the file and observe it as looking GOOD
https://i.imgur.com/qFIL5T7.png
Then once you slice it, it takes away the holes in the 4 to the right
https://i.imgur.com/K0LN55G.png

#### Project File (.3MF) where problem occurs
https://gofile.io/?c=O7ONE1
I can't attach the .3MF, Github says "does not support 3mf" so I made it a .zip and it just says it's empty...
it opens fine in CAD programs, never had this issue

In the link I have several files, all have issues. I also attached my freecad file, in freecad you can export to like 50 different formats, I put some .stl and .amf in there too.  If someone wants a different format let me know. My designs are simplified into single solid objects to make slicing smoother. As I said been doing this several  years and it just works. The holes all should work, idk why only 2 do. They're all copies of 1 design (the screw hole)

Jebtrix, 2019-11-20:
3MF files themselves are zip archives and become problematic on Linux archivers because they don't like to zip a zip. Did you just add a .zip extension to the 3MF project file? That should have been fine. Your gofile.io link is showing a service down for maintenance so may want to try another reputable share service like Mozilla's https://send.firefox.com (no account needed) and set max downloads and link duration to maximum.

Jebtrix, 2019-11-20:
I hid a couple faces on the top so you can see there is no inner hole wall geometry. Most direct way to make it manifold is to cap that top hole, which is what the slicer does.
 
[image]

rugalash, 2019-11-20:
Ok thanks. So maybe a freecad issue somehow? Just it had a bad day when drawing that part?
My friend said to redraw the circles and try it (after reading your reply) which did work for me.

rugalash, 2019-11-20:
Also thanks for the send.firefox, I just searched for a file uploader and used the first one I saw.

closed 2019-11-20 by rugalash

Jebtrix, 2019-11-20:
> Ok thanks. So maybe a freecad issue somehow? Just it had a bad day when drawing that part?
> My friend said to redraw the circles and try it (after reading your reply) which did work for me.

CAD software from free to $$$ can produce dodgy exported meshes. It's just a reality you learn to identify and workaround.