prusa3d/PrusaSlicer #3212: Specious warning about empty layers

GarthSnyder, 2019-11-20:
### Version
2.1.0+

### Operating system type + version
macOS Catalina 10.5.2 Beta (19C32e)

### 3D printer brand / version + firmware version (if known)
In point of fact, I'm set up for a Monoprice Select Mini V2. However, this issue doesn't actually involve printing.

### Behavior

The attached Star.stl file generates the error "Empty layers detected, the output would not be printable." However, orienting the object so that its flat bottom is against the build plate silences the error. It prints fine.

The fact that reorienting fixes the problem makes me suspicious that PrusaSlicer is generating the error not because of any flaw in the model but because the default orientation puts only two discrete points in contact with the build plate. I can imagine that this might result in an empty first layer. However, it's in fact not until a height of 0.61mm that PrusaSlicer generates a nonempty layer, which just seems bizarre given the geometry. Isn't there something wrong here?

[Star.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/3867098/Star.stl.zip)
[image]
[image]

lukasmatena, 2019-11-20:
> The fact that reorienting fixes the problem makes me suspicious that PrusaSlicer is generating the error not because of any flaw in the model but because the default orientation puts only two discrete points in contact with the build plate.

Your suspicion is correct. That is exactly what the check does. It basically checks every layer of the output and makes sure there is either bed or another layer below it (otherwise it would print in air).

> However, it's in fact not until a height of 0.61mm that PrusaSlicer generates a nonempty layer, which just seems bizarre given the geometry.

And what exactly seems "bizarre given the geometry"? You rested a model on conical spikes, first slices are so small that they cannot be printed. You can see all that in preview - first nonempty layer is still pretty much a point. What should the layers below it look then?

Jebtrix, 2019-11-20:
> However, it's in fact not until a height of 0.61mm that PrusaSlicer generates a nonempty layer, which just seems bizarre given the geometry. Isn't there something wrong here?

I think I understand your confusion. There is empty layers until 0.61mm, so you expect it to see an error referencing first layer height, since it technically is first empty layer. Empty layers can actually happen at any height in a project, so to the slicer it doesn't discriminate. The message references the lowest/first (printable) layer it detected an empty layer below. It is logical from the code flow point of view, the end-user not so much.

For clarity I think the empty layer message should reference Print Z layer height of the lowest/first (if consecutive) empty layer, not at the first printable layer above it where its detected from. Taking clarity a bit further the error message text could be adjusted to display "Empty Z: x.x" instead, and possibly empty layer range if applicable ie Z x.x to x.x .

GarthSnyder, 2019-11-20:
After looking into this a bit more and trying some other test shapes, I think I understand what's going on a bit better. When I originally submitted this issue, I was actually confused in several different ways. I'll explain in more detail below, but let me say up front that I don't think there's any bug here per se. However, I suspect that the inaccurate assumptions I made are probably common; I would imagine that many users in this situation would be similarly confused. The detection and presentation of this issue could be clearer and less intrusive. I'm including some suggestions below as well.

**Confusion #1:** model space vs. output space. The error message appears as soon as the model is imported, which will typically be in 3D mode, not Preview mode. This implies that there is something wrong with the model itself. The error message says "empty layer", which slicer developers no doubt correctly interpret to mean "in the rendered output, there is at least one layer with no traces". But my interpretation was more along the lines of "there is something wrong with the model that causes there to be no geometry to render for one or more layers". In fact, the error message is rather nonspecific. 

**Confusion #2:** geometry flaws vs. "too small to render" issues. Following on #1, I thought "Aha! The model is resting on two discrete points. Points have no cross-sectional area, so that must be what PrusaSlicer is complaining about". In fact, the slicer probably looks at aggregate average areas within a layer slice, so points aren't an issue at all. All of the empty layers in this case are caused by geometry cross-sections that are too small to render. But there's nothing in the workflow to clue you in to that except an oblique reference to "negligibly small extrusions".

**Confusion #3:** multiple empty layers. Following on #2, I was expecting to see a single empty first layer. So it was indeed very surprising to see empty layer after empty layer in the Preview. 0.61mm is a long way! I had no specific concept of what the minimum renderable area might be, but given that trace width was 0.45mm and there are 93 steps per mm on my (el-cheapo) machine, I would have naively expected the minimum dot size to be something like 0.45mm * 0.1mm or 0.045mm^2. But in fact, the minimum area seems to be more like 0.25mm^2, around 0.45mm * 0.45mm. I'm sure there are good reasons for that; nevertheless, it's surprising.

**Confusion #4:** no preservation of area. Here's what the cross-section of one of the object legs is at 0.44mm:

[image]

This is small, but it's certainly macroscopic and relatively compact. I would have bet money on this shape generating *some* output; its area is around 0.27mm^2, significantly larger than the minimum renderable area. But of course, PrusaSlicer doesn't see it that way. I'm guessing that it doesn't try to preserve area at all - it renders down to a grid, so elongated objects just disappear. Not unreasonable, but surprising.

**Confusion #6:** unrenderable cross-sections aren't themselves an error. PrusaSlicer deletes unrenderable cross-sections all the time without complaining about the geometry. Consider this:

[image]

A big chunk is missing from the mirrored cones, but PrusaSlicer doesn't care a bean about this because there are no completely empty layers. The inconsistency itself is confusing. 

**SUGGESTIONS**

1) There's no reason to distinguish empty layers at the top of the print from empty layers at the bottom. If you print a cone point-up (which people do all the time, see most stringing tests), there are multiple empty layers at the top, but PrusaSlicer doesn't complain. Why should multiple empty layers at the bottom be any different? Just start the print with the first nonempty layer and quit yer bitchin'. There really isn't any "error" with this star model that any user would care about.

2) Get rid of the modal dialog that pops up repeatedly after every. single. change. This should be a state indicator somewhere - perhaps in the object list (as with nonmanifold geometry), or perhaps a decoration within the Preview view. If the user goes to save, print, or transmit the G-code while there are still empty layers, a modal dialog can be opened at that point.

3) Rephrase the error report to be more explicit: "The output includes one or more layers that generated no traces. This is a problem because the empty layers will leave no base on which to print subsequent layers. In most cases, the cross-sectional geometry for the empty layers was just too small to be rendered. Inspect the empty layers in Preview mode to determine your exact situation. The first empty layer is at height 0.XXmm."

Jebtrix, 2019-11-20:
On Confusion_4, add the nozzle diameter into your illustration. Area is one thing, but dimensionally there is still "will the nozzle diameter fit inside". Only Thin Wall Detection enabled would attempt to extrude anything here.

[image]

lukasmatena, 2019-11-21:
@GarthSnyder 
> Confusion #1: Model space vs. output space. The error message appears as soon as the model is imported

No, it does not. The message appears during slicing, there is nothing to test before. You clearly have background processing on, which triggers the reslice automatically - so you of course get slicing-related error message "on import" (not just this message, you would get any other too). Turning background processing off will solve this and your suggestion number 2. And since confusion number 2 follows from 1 and confusion 3 follows from 2, this should help.

>  But my interpretation was more along the lines of "there is something wrong with the model that causes there to be no geometry to render for one or more layers".

That's why the message advices to reorient the object on the bed. And it also suggests that a faulty model be the cause, because it is impossible to recognize which of the two issues caused the empty layer.

> Confusion #3: multiple empty layers. [...]  I would have naively expected the minimum dot size to be something like 0.45mm * 0.1mm or 0.045mm^2

See @Jebtrix answer. You cannot ask for extrusion width of 0.45mm and expect PrusaSlicer to fill areas narrower than that. Lowering extrusion width may help but you are still limited by the nozzle diameter - the printer just cannot print arbitrarily narrow extrusion. It is far from trivial to plan the toolpaths for anything as big as or smaller than extrusion width (although I admit there are things to improve in PrusaSlicer - some good work related to this was done by @supermerill in his fork, we hope to be able to merge it in future).

> Confusion #6:unrenderable cross-sections aren't themselves an error

That's because unlike completely empty layer, this may not be always a show-stopper. It would also be more difficult to detect and it would increase the slicing time. So we are at least catching the most obvious scenario - an empty layer.

> Just start the print with the first nonempty layer and quit yer bitchin'

There are lots of suggestions in this issue tracker that people this we should "just" do. It is not always as easy as it seems, such a change can easily take few weeks to get right. Not to mention that blindly lowering the object can in theory produce a collision with another object. Should PrusaSlicer check for it? How long would it take? What should happen if a collision occurs? What should we "just do"? Seeing all that, we only added a check to prevent most obvious scenarios resulting in a failed print. And by the way, your model would still not print, it would detach from the bed. You'd still have to do something, such as turn on supports (which you can do now as well) or lay it flat (which is the obvious solution from the start).

lukasmatena, 2019-11-21:
@GarthSnyder 
I forgot about...

> Rephrase the error report to be more explicit

I think it says enough, but it can be author-blindness. Maybe we should rephrase it. But it cannot miss the bit warning about a possibly faulty model.