prusa3d/PrusaSlicer #3213: Reload from disk sets extruder to standard

kdsohn, 2019-11-20:
### Version
2.1.0

### Operating system type + version
WIN 10 64bit 

### Behavior
when setting a certain extruder for an object, reload from disk will reset that setting to standard instead of keeping the selected extruder
[DO6025-4-Stellplatzsender 1.zip](https://github.com/prusa3d/PrusaSlicer/files/3870239/DO6025-4-Stellplatzsender.1.zip)
#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Jebtrix, 2019-11-20:
Works as expected in latest git master so already fixed for next release.

bubnikv, 2020-03-15:
fixed in PrusaSlicer 2.2.0-rc3. Closing.

closed 2020-03-15 by bubnikv