prusa3d/PrusaSlicer #3214: Make file format output field a global option

Danafra, 2019-11-21:
### Version
2.1

### Operating system type + version
Win 10 64bit

### 3D printer brand / version + firmware version (if known)
Prusa Mk2.5

Just a feature request. It would be nice if the file format output that is currently attached to each "Print settings" preset could be moved to the preferences section of the application.  It seems to me that this should be a global setting instead of it's current state.  I guess the entry in the "Print settings" for each preset could be an override.

apphead, 2019-12-08:
I second this suggestion. I have many printer profiles and the filename should always be the same.

frank42hh, 2022-07-03:
Yes please. I want a custom format but I don't feel like having to edit several dozen printer profiles

ngroulx, 2022-11-12:
I would like to see this too. Any chance of it coming to light?

timur-tabi, 2024-08-18:
It seems dumb making this a per-print setting.  Why would I want a different file name format if printing with 0.2mm layers vs 0.15mm?  There's already a {layer_height} macro.  

Global option is okay, but I think this should be a per-printer or per-physical-printer option.  Currently, it appears that there is no macro for the nozzle width, which seems weird.  That should be fixed, too.

It's been almost four years since the bug has been opened.

jamiekin, 2025-11-23:
Adding my +1 to this issue.  Making this change to each profile is time consuming and also clutters up the UI with duplicate entries for each nozzle diameter and layer height.  There are also several people asking for this feature on Reddit.  Are there any technical reasons that this shouldn't be configurable as a global default in PrusaSlicer?

timur-tabi, 2025-11-23:
This feature should be marked as closed, because it has been fixed:

[image]

neophyl, 2025-11-23:
> This feature should be marked as closed, because it has been fixed:
> 
> <img alt="Image" width="780" height="129" src="https://private-user-images.githubusercontent.com/1065860/517887245-9bbe204f-e55c-439a-b158-4ebb6bd5c668.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjM5MzQ3NTYsIm5iZiI6MTc2MzkzNDQ1NiwicGF0aCI6Ii8xMDY1ODYwLzUxNzg4NzI0NS05YmJlMjA0Zi1lNTVjLTQzOWEtYjE1OC00ZWJiNmJkNWM2NjgucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MTEyMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTExMjNUMjE0NzM2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YjA1OGZlMTFjNWQyM2Y2Mjk3YmVlNzI3NDUxNjlmNGNhOTQwMDc3NWRjMzg0YmM2ZTdlYWQ3YWRkMjg4ZGU4ZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.39q8ds99wKtR_g6SLBt0weJdMEL_-gPaKq-WO-Icp48">

What makes you say that ?  This issue isn't about outputting in standard or binary gcode.  Its about the output file name format string located in Print Settings>Output Options.  

It most certainly is not 'fixed'.

timur-tabi, 2026-09-02:
This is still a useful feature to have.