prusa3d/PrusaSlicer #3216: [Feature Request]:  Make purge towers wider, without increasing purge volume

magicaltrevor83, 2019-11-21:
### Version
PrusaSlicer-2.1.0+win64-201909160915

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10 #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
Prusa MK3S MMU2S #s_

### Behavior
Can we make the purge towers wider - I have had a number of large prints fail where the purge tower has become unstable as it has got taller.   I know I could increase the purge volumes to achieve this but that wastes a lot of plastic
Thanks

_Is this a new feature request?_ Yes

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

lukasmatena, 2019-11-21:
It is now possible as of now. You can make the wipe tower width (Print Settings->Multiple Extruders->Wipe Tower) smaller so it ends up about rectangular and not as a narrow strip. That might be enough.

There is also a workaround taking advantage of the fact that PrusaSlicer does not check for collisions and do something like this:
[image]

I added a cylinder with no infill and use it as a support. The intersecting lines on the wipe tower print fine from our experience.

But it probably should be solved in some better way.

magicaltrevor83, 2019-11-21:
Ah, excellent idea!

I shall give that a go now

Thanks

lukasmatena, 2024-03-20:
This trick was already implemented. As of PrusaSlicer 2.7.2, wipe tower has a configurable cone-shaped support. Closing.

closed 2024-03-20 by lukasmatena