prusa3d/PrusaSlicer #3217: Crash going to preview mode when slicing for SL1

lewisj53, 2019-11-21:
### Version
_Version of PrusaSlicer used goes here_  Version 2.1.0+win64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_  Windows 10 version 1903
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

PrusaSlicer
Version:   2.1.0+win64
Build:     PrusaSlicer-2.1.0+win64-201909160915

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 18362), 64-bit edition
Total RAM size [MB]: 68,606MB
OpenGL installation
GL version:   4.6.0 NVIDIA 388.16
Vendor:       NVIDIA Corporation
Renderer:     Quadro P1000/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_cl_event
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
[... 331 lines cut ...]
GL_WIN_swap_hint
WGL_EXT_swap_control

### 3D printer brand / version + firmware version (if known)  Prusa SL1  Latest
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_

Tried to do preview slicing on a target, but it stopped and produced an error message

* _Steps needed to reproduce the problem_ Try to enter preview mode.
  * _If this is a command-line slicing issue, include the options used_ N/A
* _Expected Results_ Reviewable sliced piece
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
There is no image of the piece on the platform.

_Is this a new feature request?_  No

[MANITOU_04.zip](https://github.com/prusa3d/PrusaSlicer/files/3873812/MANITOU_04.zip)

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

lukasmatena, 2019-11-21:
> but it stopped and produced an error message

What error message? It will be hard to debug if you don't share that.

It could be something related to the graphics driver. Could you run PrusaSlicer from the console as `prusa-slicer-console.exe --sw-renderer --loglevel=9` and check whether it still crashes (and if it does, paste the last lines from the log in the console)?

lewisj53, 2019-11-21:
I have a notion that the problem is that the bed, that is made by the feet of the supports, actually crosses into no-man’s land in some places and the algorithm stops because of it.

I will try to get a screenshot of the error message.

Lewis.

From: lukasmatena
Sent: Thursday, 21 November 2019 21:10
To: prusa3d/PrusaSlicer
Cc: lewisj53; Author
Subject: Re: [prusa3d/PrusaSlicer] Crash going to preview mode when slicingfor SL1 (#3217)

but it stopped and produced an error message
What error message? It will be hard to debug if you don't share that.
It could be something related to the graphics driver. Could you run PrusaSlicer from the console as prusa-slicer-console.exe --sw-renderer --loglevel=9 and check whether it still crashes (and if it does, paste the last lines from the log in the console)?

lewisj53, 2019-11-21:
Here is the error message.

I might be wrong about the pad overstepping the edge causing the problem.

From: lukasmatena
Sent: Thursday, 21 November 2019 21:10
To: prusa3d/PrusaSlicer
Cc: lewisj53; Author
Subject: Re: [prusa3d/PrusaSlicer] Crash going to preview mode when slicingfor SL1 (#3217)

but it stopped and produced an error message
What error message? It will be hard to debug if you don't share that.
It could be something related to the graphics driver. Could you run PrusaSlicer from the console as prusa-slicer-console.exe --sw-renderer --loglevel=9 and check whether it still crashes (and if it does, paste the last lines from the log in the console)?

lewisj53, 2019-11-22:
Here is a log file.

The thing seemed to crash before I could demonstrate the problem I wrote about.
I will try again.

Lewis.

From: lukasmatena
Sent: Thursday, 21 November 2019 21:10
To: prusa3d/PrusaSlicer
Cc: lewisj53; Author
Subject: Re: [prusa3d/PrusaSlicer] Crash going to preview mode when slicingfor SL1 (#3217)

but it stopped and produced an error message
What error message? It will be hard to debug if you don't share that.
It could be something related to the graphics driver. Could you run PrusaSlicer from the console as prusa-slicer-console.exe --sw-renderer --loglevel=9 and check whether it still crashes (and if it does, paste the last lines from the log in the console)?

lewisj53, 2019-11-22:
Here is another log.

The error in question appeared this time

Lewis.

From: lukasmatena
Sent: Thursday, 21 November 2019 21:10
To: prusa3d/PrusaSlicer
Cc: lewisj53; Author
Subject: Re: [prusa3d/PrusaSlicer] Crash going to preview mode when slicingfor SL1 (#3217)

but it stopped and produced an error message
What error message? It will be hard to debug if you don't share that.
It could be something related to the graphics driver. Could you run PrusaSlicer from the console as prusa-slicer-console.exe --sw-renderer --loglevel=9 and check whether it still crashes (and if it does, paste the last lines from the log in the console)?

lewisj53, 2019-11-22:
Here is another.  In this instance, it completed the supports, although it produced the statement, “An object outside the print area was detected.  Resolve the current problem to continue slicing”.  This statement was not there before the automatic support procedure was started.

Lewis

From: lukasmatena
Sent: Thursday, 21 November 2019 21:10
To: prusa3d/PrusaSlicer
Cc: lewisj53; Author
Subject: Re: [prusa3d/PrusaSlicer] Crash going to preview mode when slicingfor SL1 (#3217)

but it stopped and produced an error message
What error message? It will be hard to debug if you don't share that.
It could be something related to the graphics driver. Could you run PrusaSlicer from the console as prusa-slicer-console.exe --sw-renderer --loglevel=9 and check whether it still crashes (and if it does, paste the last lines from the log in the console)?

rtyr, 2021-02-04:
Is this issue still valid? Does it happen with PS2.3.0?

closed 2021-09-23 by bubnikv