prusa3d/PrusaSlicer #3218: Cannot reach/see the top of a tall piece

lewisj53, 2019-11-21:
### Version
_Version of PrusaSlicer used goes here_

Version 2.1.0+win64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
PrusaSlicer
Version:   2.1.0+win64
Build:     PrusaSlicer-2.1.0+win64-201909160915

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 18362), 64-bit edition
Total RAM size [MB]: 68,606MB
OpenGL installation
GL version:   4.6.0 NVIDIA 388.16
Vendor:       NVIDIA Corporation
Renderer:     Quadro P1000/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_cl_event
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
[... 335 lines cut ...]
GL_WIN_swap_hint
WGL_EXT_swap_control

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_  Yes

The image is as small as it can be.  If I want to rotate in the x-y plane, I cannot reach the arrow handle because it is at the top of the screen and obscured by the controls there.

1.  It would be good if the level of the platform could be lowered on the screen.
2.  It would be good to have a control that sets the three planes,  x-y, y-z and x-z, flat on the screen so that sizes and alignments can be judged by eye.

[MANITOU_04.zip](https://github.com/prusa3d/PrusaSlicer/files/3873868/MANITOU_04.zip)

[image]

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

jmd73, 2019-11-21:
Press and hold the right mouse button in the grey area outside the object and drag the mouse to move the whole printbed.

lewisj53, 2019-11-21:
Thanks.

From: jmd73
Sent: Thursday, 21 November 2019 20:44
To: prusa3d/PrusaSlicer
Cc: lewisj53; Author
Subject: Re: [prusa3d/PrusaSlicer] Cannot reach/see the top of a tall piece(#3218)

Press and hold the right mouse button in the grey area outside the object and drag the mouse to move the whole printbed.

bubnikv, 2020-03-15:
is the issue solved?

bubnikv, 2020-03-26:
no response, closing.

closed 2020-03-26 by bubnikv