prusa3d/PrusaSlicer #3219: Feature Request: Hollow Holes and remove internal support

debk3141, 2019-11-21:
I'd like to see Prusa slicer able to hollow out an object and ability to place holes in an objects.  I have been using a program called Asura 3d, that came with the Peoploy Moai to hollow out and place holes and then export the result as a stl to slice in Prusa Slicer.  

It would  be nice to be able to do all of this in Prusa Slicer.

Also an option to remover internal supports on hollowed out objects would be appreciated

 

Debra

lukasmatena, 2019-11-21:
We are currently working on object hollowing and for placing the drain holes for SLA printing. Both will (probably) be in PrusaSlicer2.2.0.

debk3141, 2019-11-21:
Great! , Thanks for the quick response.  I look forward to this

Debra

bubnikv, 2020-05-15:
Hollowing and drainage holes implemented in PrusaSlicer 2.2.0.

> Also an option to remover internal supports on hollowed out objects would be appreciated

Supports from build plate only.

Closing.

closed 2020-05-15 by bubnikv