prusa3d/PrusaSlicer #3220: Support for tubelanes

scooper4711, 2019-11-21:
Feature request: recent scientific research has found a stronger print pattern. It would be very cool if we could print with this pattern as infill.
https://www.fastcompany.com/90433748/were-on-the-verge-of-making-buildings-as-strong-as-diamonds?partner=rss&utm_campaign=rss+fastcompany&utm_content=rss&utm_medium=feed&utm_source=rss

Jebtrix, 2019-11-21:
Interesting. Here is a more detailed link to the actual research paper with breakdown illustrations.

https://www.researchgate.net/publication/320035893_Multi-scale_Geometric_Design_Principles_Applied_to_3D_Printed_Schwarzites

scalablecory, 2019-11-21:
Here's an interesting quote:

> All the tested materials are stronger when printed in these shapes than they were as solid blocks.

closed 2026-09-29 by prusaslicer-bot[bot]