prusa3d/PrusaSlicer #3223: Decrease layer height if it takes the same time

espr14, 2019-11-21:
v2.1.0 win64

Printer can print fast with higher temperature but filament has to melt fast enough. For such issue there's a "Max volumetric speed" option. If the layer area is too small the print is slowed down. If the speed is limited it would be better to decrease layer height instead. It would bring more details in the same print time. Can you add such option which would dynamicaly decrease layer height so it prints fast still?

pepelevamp, 2019-11-25:
smart

espr14, 2019-11-26:
Printing speed is also limited by cooling speed so this should be taken into account as well.

Smaller layer height always takes longer because of non-printing movements. However, user can tolerate x% (1%) time added if it brings y% (10%) smaller layers.

The solution is hard and even harder if you take variable layer height into account.

Kachidoki2807, 2019-11-26:
Brillant idea!
It could even go further, by increasing the layer height if the max volumetric speed is not reached. I often increase manually the layer height (and/or the perimeters speed) to speed up the print of prototypes. I also sometimes uncheck the "auto cooling" to go even faster when prototypes can be ugly.
In fact, it could be summarized by: "doing the best to achieve the highest quality (or speed)" if min layer time (or max volumetric speed is not reached) is reached.

espr14, 2019-11-26:
Currently, cooling speed is fixed so it is not proportional to mass of filament. This should be improved as well otherwise it will penalize smaller layer heights.

closed 2026-09-29 by prusaslicer-bot[bot]