prusa3d/PrusaSlicer #3224: Extruder temperature taken from first extruder when printing single material on second extruder

gnydick, 2019-11-22:
### Version
2.1.0+win64

### Operating system type + version
Windows 10 1909 64-bit
### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* I have 2 extruders
* I print with one material, but using the second extruder
* the temperature for the hot-end is taken from whatever filament is assigned to the first extruder

#### Project File (.3MF) where problem occurs
[tv dinner table foot v2.zip](https://github.com/prusa3d/PrusaSlicer/files/3877642/tv.dinner.table.foot.v2.zip)

lukasmatena, 2019-11-22:
Your custom start gcode says:

`M109 S[first_layer_temperature_0]`

So you explicitly demand temperature from the first filament. In other words, PrusaSlicer would work fine if you didn't tell it not to. Remove the "_0" and it should work.

rtyr, 2021-02-04:
Answered. No response. Closing.

closed 2021-02-04 by rtyr