prusa3d/PrusaSlicer #3226: assert fail when generating gyroid infill - 3mf attached

BeldrothTheGold, 2019-11-22:
### Version
_Version of PrusaSlicer used goes here_
git version_2.1.0-513-g3b084c50c

### Operating system type + version
tested in Manjaro Linux (g++ 9.2) and in Windows 10 (MSVC 2019)

### Behavior
reproducible assert fail or segfault when generating infill for attached 3mf

```(prusa-slicer:26871): Gtk-CRITICAL **: 03:11:22.938: IA__gtk_tree_row_reference_new_proxy: assertion 'path->depth > 0' failed
prusa-slicer: /home/scott/dev/git/PrusaSlicer/src/libslic3r/Fill/../EdgeGrid.hpp:96: void Slic3r::EdgeGrid::Grid::visit_cells_intersecting_line(Slic3r::Point, Slic3r::Point, VISITOR&) const [with VISITOR = Slic3r::mark_boundary_segments_touching_infill(const std::vector<std::vector<Slic3r::Point> >&, std::vector<std::vector<Slic3r::ContourPointData> >&, const Slic3r::BoundingBox&, const Polylines&, double, double)::Visitor]: Assertion `iy >= 0 && size_t(iy) < m_rows' failed.
```

#### Project File (.3MF) where problem occurs
[dice_tower_v3.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3879017/dice_tower_v3.3mf.zip)

BeldrothTheGold, 2019-11-22:
call stack:
```
libc.so.6!raise (Unknown Source:0)
libc.so.6!abort (Unknown Source:0)
libc.so.6!__assert_fail_base.cold (Unknown Source:0)
libc.so.6!__assert_fail (Unknown Source:0)
Slic3r::EdgeGrid::Grid::visit_cells_intersecting_line<Slic3r::mark_boundary_segments_touching_infill(const std::vector<std::vector<Slic3r::Point> >&, std::vector<std::vector<Slic3r::ContourPointData> >&, const Slic3r::BoundingBox&, const Polylines&, double, double)::Visitor>(Slic3r::Point, Slic3r::Point, Slic3r::Visitor &) const(const Slic3r::EdgeGrid::Grid * const this, Slic3r::Point p1, Slic3r::Point p2, Slic3r::Visitor & visitor) (/home/scott/dev/git/PrusaSlicer/src/libslic3r/EdgeGrid.hpp:96)
Slic3r::mark_boundary_segments_touching_infill(const std::vector<std::vector<Slic3r::Point, std::allocator<Slic3r::Point> >, std::allocator<std::vector<Slic3r::Point, std::allocator<Slic3r::Point> > > > & boundary, std::vector<std::vector<Slic3r::ContourPointData, std::allocator<Slic3r::ContourPointData> >, std::allocator<std::vector<Slic3r::ContourPointData, std::allocator<Slic3r::ContourPointData> > > > & boundary_data, const Slic3r::BoundingBox & boundary_bbox, const Slic3r::Polylines & infill, const double clip_distance, const double distance_colliding) (/home/scott/dev/git/PrusaSlicer/src/libslic3r/Fill/FillBase.cpp:820)
Slic3r::Fill::connect_infill(Slic3r::Fill * const this, Slic3r::Polylines && infill_ordered, const Slic3r::ExPolygon & boundary_src, Slic3r::Polylines & polylines_out, const Slic3r::FillParams & params) (/home/scott/dev/git/PrusaSlicer/src/libslic3r/Fill/FillBase.cpp:906)
Slic3r::FillGyroid::_fill_surface_single(Slic3r::FillGyroid * const this, const Slic3r::FillParams & params, unsigned int thickness_layers, const std::pair<float, Slic3r::Point> & direction, Slic3r::ExPolygon & expolygon, Slic3r::Polylines & polylines_out) (/home/scott/dev/git/PrusaSlicer/src/libslic3r/Fill/FillGyroid.cpp:198)
Slic3r::Fill::fill_surface(Slic3r::Fill * const this, const Slic3r::Surface * surface, const Slic3r::FillParams & params) (/home/scott/dev/git/PrusaSlicer/src/libslic3r/Fill/FillBase.cpp:75)
Slic3r::Layer::make_fills(Slic3r::Layer * const this) (/home/scott/dev/git/PrusaSlicer/src/libslic3r/Fill/Fill.cpp:375)
Slic3r::PrintObject::<lambda(const tbb::blocked_range<long unsigned int>&)>::operator()(const tbb::blocked_range<unsigned long> &) const(const Slic3r::PrintObject::<lambda(const tbb::blocked_range<long unsigned int>&)> * const __closure, const tbb::blocked_range<unsigned long> & range) (/home/scott/dev/git/PrusaSlicer/src/libslic3r/PrintObject.cpp:382)
tbb::interface9::internal::start_for<tbb::blocked_range<long unsigned int>, Slic3r::PrintObject::infill()::<lambda(const tbb::blocked_range<long unsigned int>&)>, const tbb::auto_partitioner>::run_body(tbb::blocked_range<unsigned long> &)(tbb::interface9::internal::start_for<tbb::blocked_range<long unsigned int>, Slic3r::PrintObject::infill()::<lambda(const tbb::blocked_range<long unsigned int>&)>, const tbb::auto_partitioner> * const this, tbb::blocked_range<unsigned long> & r) (/usr/include/tbb/parallel_for.h:115)
tbb::interface9::internal::dynamic_grainsize_mode<tbb::interface9::internal::adaptive_mode<tbb::interface9::internal::auto_partition_type> >::work_balance<tbb::interface9::internal::start_for<tbb::blocked_range<long unsigned int>, Slic3r::PrintObject::infill()::<lambda(const tbb::blocked_range<long unsigned int>&)>, const tbb::auto_partitioner>, tbb::blocked_range<long unsigned int> >(tbb::interface9::internal::start_for<tbb::blocked_range<long unsigned int>, Slic3r::PrintObject::infill()::<lambda(const tbb::blocked_range<long unsigned int>&)>, const tbb::auto_partitioner> &, tbb::blocked_range<unsigned long> &)(tbb::interface9::internal::dynamic_grainsize_mode<tbb::interface9::internal::adaptive_mode<tbb::interface9::internal::auto_partition_type> > * const this, tbb::interface9::internal::start_for<tbb::blocked_range<long unsigned int>, Slic3r::PrintObject::infill()::<lambda(const tbb::blocked_range<long unsigned int>&)>, const tbb::auto_partitioner> & start, tbb::blocked_range<unsigned long> & range) (/usr/include/tbb/partitioner.h:438)
tbb::interface9::internal::partition_type_base<tbb::interface9::internal::auto_partition_type>::execute<tbb::interface9::internal::start_for<tbb::blocked_range<long unsigned int>, Slic3r::PrintObject::infill()::<lambda(const tbb::blocked_range<long unsigned int>&)>, const tbb::auto_partitioner>, tbb::blocked_range<long unsigned int> >(tbb::interface9::internal::start_for<tbb::blocked_range<long unsigned int>, Slic3r::PrintObject::infill()::<lambda(const tbb::blocked_range<long unsigned int>&)>, const tbb::auto_partitioner> &, tbb::blocked_range<unsigned long> &)(tbb::interface9::internal::partition_type_base<tbb::interface9::internal::auto_partition_type> * const this, tbb::interface9::internal::start_for<tbb::blocked_range<long unsigned int>, Slic3r::PrintObject::infill()::<lambda(const tbb::blocked_range<long unsigned int>&)>, const tbb::auto_partitioner> & start, tbb::blocked_range<unsigned long> & range) (/usr/include/tbb/partitioner.h:256)
tbb::interface9::internal::start_for<tbb::blocked_range<long unsigned int>, Slic3r::PrintObject::infill()::<lambda(const tbb::blocked_range<long unsigned int>&)>, const tbb::auto_partitioner>::execute(void)(tbb::interface9::internal::start_for<tbb::blocked_range<long unsigned int>, Slic3r::PrintObject::infill()::<lambda(const tbb::blocked_range<long unsigned int>&)>, const tbb::auto_partitioner> * const this) (/usr/include/tbb/parallel_for.h:142)
libtbb_debug.so.2![Unknown/Just-In-Time compiled code] (Unknown Source:0)
libpthread.so.0!start_thread (Unknown Source:0)
libc.so.6!clone (Unknown Source:0)
```

lukasmatena, 2019-11-22:
Thanks for the report. This part of code is still under significant development by @bubnikv, maybe he already knows about it. It will be fixed.

BeldrothTheGold, 2019-11-22:
When the assertion fails: 

[image]

BeldrothTheGold, 2019-11-22:
> Thanks for the report. This part of code is still under significant development by @bubnikv, maybe he already knows about it. It will be fixed.

Understood! Thanks

bubnikv, 2019-11-22:
The new infill path planning algorithm has a bug. Thanks for pointing out.

bubnikv, 2019-11-22:
It should be fixed with 9c4dc80057dd311b2fee4c639918f0cf0b5642a1
Thanks for heads up.

closed 2019-11-22 by bubnikv

BeldrothTheGold, 2019-11-22:
Indeed fixed. Thanks!

commit 9c4dc800 (2019-11-22): Fix of the new gyroid infill path planning. Fixes #3226