prusa3d/PrusaSlicer #3228: Access violation / crash when slicing

F0x06, 2019-11-22:
Commit: 232f3249990143709410fc02160fbaecb923baff
Version: `2.1.0`
Platform: `Win64`

PrusaSlicer crashes when slicing Lack enclosure batch 7 (.3mf attached)

Note: PrusaSlicer is not crashing when compiled in debug mode.

```
ShortestPath.cpp:691
```

```
Exception thrown at 0x00007FFF147F247E (PrusaSlicer.dll) in prusa-slicer-console.exe: 0xC0000005: Access violation reading location 0xFFFFFFFFFFFFFFFF.
```

 
[prusa_mmu_enclosure_7_issue.zip](https://github.com/prusa3d/PrusaSlicer/files/3880301/prusa_mmu_enclosure_7_issue.zip)

Jebtrix, 2019-11-22:
Slices fine now with commit https://github.com/prusa3d/PrusaSlicer/commit/2b17e81f133dbc2aeb22d99d857b6a1d136ea247

F0x06, 2019-11-22:
@Jebtrix I confirm this works, i'll close the issue thanks ! :)

closed 2019-11-22 by F0x06