prusa3d/PrusaSlicer #3230: 3D-Viewport low performance/fps

F0x06, 2019-11-22:
Commit: 2b17e81f133dbc2aeb22d99d857b6a1d136ea247
Version: `2.1.0`
Platform: `Win64`
GPU: `2080 Ti`

When using the latest release (version_2.1.0) i have full 144 fps in 3D/preview
On the latest commit 2b17e81f133dbc2aeb22d99d857b6a1d136ea247 i have around 5-15 fps

Maybe you are working on something, i'll report it in case of :)

Thanks.

enricoturri1966, 2019-11-25:
This slow down is due to 3Dconnexion devices detection. Working on it to improve performance.

F0x06, 2019-11-25:
Fixed by commit ab00f501f1363802282986ea9ab86d2e305bc24d, thanks for the reactivity and quick response ! :)

closed 2019-11-25 by F0x06

Kachidoki2807, 2019-11-26:
> This slow down is due to 3Dconnexion devices detection. Working on it to improve performance.

Do you mean the 3Dconnexion devices are already implemented into the 2.1.0?
I don't think so, or maybe I missed something?

enricoturri1966, 2019-11-26:
@Kachidoki2807
They are supported in current master, not in any released version.