prusa3d/PrusaSlicer #3232: Feature Request: "Avoid Crossing External Perimeters" Only

Sylkun, 2019-11-24:
Hi,
Please, add a sub-option "External Perimeters only" to the existing "Avoid Crossing Perimeters" option, to allow slicer crossing internal perimeters and prevent the behavior below:
(external in orange, internal in yellow)
[image]

Thanks.
Cheers

PS: I'm using PrusaSlicer 2.1.0+win64-201909160915

bubnikv, 2020-12-11:
"Avoid crossing perimeters" has been reworked with 5f456ebbe78f3b994ea7ecc7de23c119e4705b61 (see the branch lh_avoid_crossing_perimeters).
The new algorithm will be available in PrusaSlicer-2.3.0-beta3. Please check the PrusaSlicer-2.3.0-beta3 change log for details.

We would like to test with your model. Would you please provide it?

rtyr, 2021-09-20:
No response. Closing.

closed 2021-09-20 by rtyr