prusa3d/PrusaSlicer #3233: MMU2S 0.25mm nozzle (feature request), wipe tower damage and under extrusion

bartlee01, 2019-11-24:
### Version
2.1.0+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa i3 Mk3S MMU2S
3.8.1 and 1.0.6

### Behavior
With 0.4mm nozzle in MMU2S mode with wipe tower I have to make sure the first 3 layers have no colour change, or the wipe tower will be torn up and stick to the nozzle.

When leaving the wipe tower the nozzle is not primed so the first couple of cm of print does not work. I have found a work around by having 2 perimeters of skirt the height of the last colour change layer and if you do this then for the first layer skirt you have to increase the minimum extrusion amount until it slices with 2 perimeters.

Also, I am using the 0.4mm nozzle to print multicoloured buttons and jewelry. A resin printer is not suitable as it can't print multicolour. I would really like it if Prusa introduced support for 0.25mm nozzle and MMU2S.

supermerill, 2019-11-25:
> I would really like it if Prusa introduced support for 0.25mm nozzle and MMU2S.

It's not possible to ram the (extruding very quickly) all the melted plastic that lie in the hot zone of the extruder via a 0.25 nozzle. So you can't have a nice tip, so the mmu will jam too frequently.
=> so no 0.25 nozzle profile.

(just repeating what I remember from an other issue thread)

If you really want 0.25mm multi-colour, maybe you have to invest in a tool-changer printer.

rtyr, 2020-04-22:
Your issues with the wipe tower are most likely not PrusaSlicer related. Maybe wrong IR sensor calibration, wrong profile selected (MMU2 vs 2S), bed preparation, liveZ? You can contact tech-support about this.

About 0.25mm nozzle. It is basically like supermerill explained. 

If you are still experiencing these issues with 2.2.0, please provide 3MF project file so we can test.

rtyr, 2020-05-09:
No response. Closing.

closed 2020-05-09 by rtyr