prusa3d/PrusaSlicer #3234: Increasing Purging Volume on Extruder 2 Causes Unwanted Movements

cosmith71, 2019-11-24:
### Version
2.1.0

### Operating system type + version
Windows 10

### Behavior
Increasing the extruder purging volume on extruder 2 to 90 or greater (89 seems fine) creates unwanted extruder movements in the form of lines at the front of the bed.  This does not seem to be the case with any other extruder.  It does not show on the sliced plater view but it is present when the generated gcode is rendered externally.

Renders from Repetier-Server.

Extruder 2 purge volume at 89:

[image]

Extruder 2 purge volume at 90:

[image]

Render with purge 2 set to 90 from online viewer gcode.ws:

[image]

.3mf file:

[3C_Lizard_1.zip](https://github.com/prusa3d/PrusaSlicer/files/3883692/3C_Lizard_1.zip)

cosmith71, 2019-11-24:
It appears to affect extruder 5 as well.

cosmith71, 2019-11-24:
It appears to involve more than 2 and 5, although I cannot nail down a pattern.

lukasmatena, 2019-11-26:
Did you actually print it? If you did, have you seen the extra extrusions? Or do they only appear in the gcode preview?

jmd73, 2019-11-26:
Can this be related to  issue #2968 that I created, and Bubnik closed with a rather cryptic reply?
An additional note to my issue that I uncovered later is that if I have all default settings with 0.2mm settings and change the layer hight to 0.201mm the extra movements dissapears.

lukasmatena, 2019-11-26:
@jmd73 The reply is cryptic because it erroneously mentions 2.0.0 instead of 2.1.0. The close was justified though. The bug reported in #2968 passed through testing and almost made it into the 2.1.0 release. You discovered and we fixed it just before the release as far as I remember. If you still see that issue in 2.1.0 final or current master, it should be reopened - let us know there. But think that one was too serious to not be discovered by anyone else.

I am not sure if this issue is related or not. But first of all we need to be sure it is actually PrusaSlicer's fault. I already debugged several issues that turned out to only show in Repertier preview and were not actually in the gcode.

jmd73, 2019-11-26:
@lukasmatena I did build from source. (As seen on screenshots.) Last time I tried was this friday, and I had the same result. The report was after the release. I erroneously only took screencaps of the built version. The 2.1.0 release does the same, and those screens should have been captured.

I you can try my .3mf that I attached and don't get the extra travel/purge moves, I will try a clean build again. But the result is the same in the released 2.1.0.

cosmith71, 2019-11-26:
@lukasmatena I'm not sure if any plastic was extruded but it did make the extra moves at a very slow speed (the change in sound is what tipped me off).  I stopped it at that point.

jmd73, 2019-11-26:
@lukasmatena And as I stated in my report. I actually got it IRL (i.e. while printing.) I can take a photo of the failed print.

And as @cosmith71 writes above, it's at a slower speed. And it extruded plastic.

The failed print was scaled up so the line I got was thru the object.

I only use 2.0.0 for mmu prints now, because of this.

lukasmatena, 2019-11-26:
@jmd73 
> And as I stated in my report. I actually got it IRL (i.e. while printing.)

I know, I read it. My question about whether it is just a gcode preview issue was meant for @cosmith71, because this is his issue and although this can have the same cause as yours, we need to be sure first. Until then, #2968 should be discussed in #2968 (a reference from one to the other is of course welcome).

If you're saying that #2968 was not really fixed, something is wrong. I will look into both issues when I have time.

lukasmatena, 2019-12-05:
There really was a bug regarding the move to the wipe tower. This should now be fixed by 
4ba7cb7 as @jmd73 confirmed in #2968. This issue is very likely to have the same cause. The fix will be in upcoming PrusaSlicer 2.2 release.

Closing the issue, thanks for the report.

closed 2019-12-05 by lukasmatena

cosmith71, 2019-12-05:
Awesome!  Thanks!