prusa3d/PrusaSlicer #3235: Extra fan time

espr14, 2019-11-25:
v2.1.0

Fan needs some time to spin up and down. Moreover, some FWs do not put M106 into the movement speed so the fan is not synchronized with printing.

Extra time could be added or it could be shifted by specific number of instructions so the fan starts sooner/later and stops sooner/later.

supermerill, 2019-11-26:
implemented in my fork, commit supermerill/slic3r@c6239850b084845556c524c0d41df7d5f89b009a

pKai3, 2024-03-03:
Hi, just wondering if there are any plans to implement this in a current version... Without having looked into any details I blindly suggest that one might be able to fork this feature off of some of the work implemented by pressure advance... Please let me know if there's anything I can do to help advance this :)

Thanks!

Tupson444, 2026-09-12:
still relevant