prusa3d/PrusaSlicer #3236: PrusaSlicer crashes when using layer slider up/down on certain model

Boby71, 2019-11-25:
### Version
2.10 WIN10/64Bit

### Operating system type + version
Win10/64Bit V1903

### 3D printer brand / version + firmware version (if known)
MK2.5s - latest firmware

### Behavior
When I load the attached model into the slicer and move the slider in the sliced preview up/down a few times, I get the attached bug report. Also there is a strange "jump" in displaying the sliced model when slowly sliding up from the bottom upwards. I guess the model itself is a bit faulty (was imported into Fusion360, slightly modified and exported as STL), but this should not result into such kind of error message. The repair through Netfabb didn't solve the "jump" problem.

[image]

_Is this a new feature request?_
Nope

#### Project File (.3MF) where problem occurs
[XMAS Tree Modern V2 v4.zip](https://github.com/prusa3d/PrusaSlicer/files/3889085/XMAS.Tree.Modern.V2.v4.zip)

Jebtrix, 2019-11-26:
I can't reproduce the crash. The reason it jumps is because of the bottom where there is a hole. With the model the way it is it can't actually do vase mode until it is above that feature. Around the center hole feature there is two disconnected features, the hole and the outer shell that prevents vase.

JohnnyDeer, 2020-03-16:
Hi, i'm also not able to reproduce it. Is there some specific place in preview where it crashing? Could you please try to reproduce it on version 2.2.0?

bubnikv, 2020-10-17:
no response, closing.

closed 2020-10-17 by bubnikv