prusa3d/PrusaSlicer #3237: Firmware can not be updated

gebuwa, 2019-11-26:
### Version
_Version of PrusaSlicer used goes here_
Prusa3D_win_2_2_8

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version  Windows 7 / 64bit
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)  aktuell 3.8.0
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
The flashing of the firmware can not be started, communication problem. It is only available as "COM1" or LPT1 even though the printer is connected to the USB port.

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

[image]

_Is this a new feature request?_
Yes

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2020-03-15:
most likely you don't have the printer USB driver installed.

rtyr, 2020-04-22:
@gebuwa Is it solved? Can we close this?

gebuwa, 2020-04-24:
Ok, can be closed.
With best regards, "gebuwa"

 

 

 

Gesendet: Mittwoch, 22. April 2020 um 13:35 Uhr

closed 2020-04-24 by rtyr