prusa3d/PrusaSlicer #3238: Opening a .3mf Project Doesn't Recognize Existing Printer Setting Profile

sgano, 2019-11-26:
### Version
v2.1.0

### Operating system type + version
MacOS 10.15.1

### 3D printer brand / version + firmware version (if known)
MK3S

### Behavior
When opening a saved .3mf project file - it recognizes the proper "Print Setttings" profile that was used and selects it in the interface, it also does the same for "Filament", but the "Printer" is always titled as the file name then in parenthesis shows the original printer profile.  What I would expect is that the proper Printer profile would be selected -- and any differences made to that profit should be highlighted like usual -- it shouldn't show up as a completely new Printer profile since it was saved with an existing one.    This has a lot of ramifications such as it doesn't seem to save the Octoprint host or API (which means I can't re-upload the file directly to Octoprint when re-opening a file).  It also is just confusing to have a new Printer profile appear in the interface the entire session until I close and reopen the program -- when it should really just inherit from the proper profile that was used when the file was created and saved.  

#### Project File (.3MF) where problem occurs
occurs with all .3MF files.

neophyl, 2019-11-26:
The problem is that the profile on your computer could have been modified in the intervening time.  That is why is saves a complete set of all parameters and appends the name that was used.  Saving a copy internally of the profiles means that its easy to transfer a entire project from one machine to another.  Its extremely useful tool for sharing and debugging prints.   Its easy just to reset it to the current version of that profile. 
Personally I would not be a fan if the behaviour was changed.

Kachidoki2807, 2019-11-26:
I agree with @neophyl, but I can propose an intermediate solution.
If the parameters are identical, by comparing them or merely doing a hash of the whole settings, PrusaSlicer could display the existing profile instead of the imported one.
It is purely cosmetic, but nonetheless important for the user experience.

supermerill, 2019-11-26:
> If the parameters are identical, by comparing them or merely doing a hash of the whole settings, PrusaSlicer could display the existing profile instead of the imported one.

It isn't already the case, no? I just test it with a filament and it seems to work.

sgano, 2019-11-26:
Good points.  I agree the full set of settings should be stored so it can be used universally, but like what @Kachidoki2807 mentioned if the orignal profile name exists it should be selected, and then diffed to show the differences between the settings stored in the .3mf and the local profile.  That way even if the local profile has changed, it simply shows the diff from the current version and the file.      For the example of the Octopi settings -- that may be a special case as I can understand not saving that with the file and therefore it would always be a diff.  Maybe that setting should be more universal in some way.

sgano, 2019-11-26:
@supermerill -- make a change to a setting in the Filament (e.g., a temperature) then save the file, close the app and reopen the file.  It won't select the original profile.

supermerill, 2019-11-26:
@sgano because now the  parameters are not identical.
I was responding to @Kachidoki2807 , saying that i think that his proposal is already implemented.

Kachidoki2807, 2019-11-26:
> It's already the case, no? I just test it with a filament and it seems to work.
Indeed, my apologies, I always saw all the 3 profiles (print/filament/printer) imported from the .3mf, probably because I always save the .3mf project when I do so much changes. I tried to save a project with a modified profile, and then restore back the standard profile and re-save it. At the re-opening it selected correctly the standard profile.

This is a correct behavior from my point of view.

The only way to trap it is to modify a parameter, save and reopen the project, and then undo manually the modification and resave and reopen the project. In this case, all the parameters are identical, but the profile is not recognized because the name is wrong, prefixed the .3mf filename. And moreover, if you repeat this scenario, the prefix is added once more.

@sgano the behavior you describe is already implemented, excepted for the profile name which is put between parenthesis and prefixed by the .3mf filename. But the diff with the original profile is present.

sgano, 2019-11-26:
Ah I see that now, thanks.  Though I still think instead of the filename in the profile it should appear as "Original profile (modified)" like it does when you create/save the file  if that profile is present locally instead of using the filename  -- that filename profile even remains in the UI even after starting a new project without closing the application.  I am ending up with a lot of nested filenames in the profiles, which makes it rather messy/confusing in the UI).

sgano, 2019-11-26:
One final comment:  the "Printer Settings" profile is never automatically selected even if changes are not made to that saved user profile unlike the Filament Settings and Print Settings.  At the very least it would seem that if no changes are made to the Printer Settings, it should be automatically selected instead of appending the filename to the front of the printer settings name -- to be consistent with how the other profiles work.  (this would at least solve my issue with Octoprint settings not being restored when opening a file #3244)

bubnikv, 2020-03-16:
Just to summarize: The problem seems to be:

1) 3mf with Prusament PLA created
2) 3mf reopened, print temp changed, 3mf stored
3) 3mf reopened, the print profile is shown as "project name.3mf (Prusament PLA)"
4) print temp changed back to the system profile value, 3mf stored.
5) 3mf reopened, the print profile is shown as "project name.3mf (project name.3mf (Prusament PLA))

I suppose in this case we may change the name back to "Prusament PLA".

The same will happen if at the point 4) the temp is changed to some value different from the system settings. What do you propose then?

sgano, 2020-04-02:
Yes, I think you are right in step 5 in that case it should revert back to Prusament PLA if there is no longer any difference from the original profile. 

However, at step four (or even at a step 6) -- if the value (or any value) is again changed to not match all of the settings then just go back to the "project name.3mf (Prusament PLA)" convention. 

Though if the project name changes there could be some merit to having all of those nested project names in the profile.  But I am not sure how useful that is.  Just having a single depth (in this case:  "project name.3mf (Prusament PLA)"  or "Prusament PLA")  seems like it should be enough, and would limit how long those names could get.

rtyr, 2021-09-05:
Should be fixed since 2.3.0. Closing.

closed 2021-09-05 by rtyr